esp-rs/esp-hal #1698: How to override the default post_init function?

konsulten, 2024-06-19:
Currently it seems the watchdogs are disabled by default in the __post_init function. Is there a possibility to override or modify this behaviour?

https://github.com/esp-rs/esp-hal/pull/763

Dominaezzz, 2024-06-19:
I'm pretty sure you can just re-enable the watchdog in your main function. There should be an example as well.

konsulten, 2024-06-19:
Won’t there be some time where its disabled then? I would prefer never to
disable it

MabezDev, 2024-06-19:
We'd probably add this behind a configuration option. We don't have a complete story for this yet see #1111.

Dominaezzz, 2024-06-19:
For this sort of thing, rather than a config option, which can only be specified once per crate. This should probably be part of the entry macro, which can be specified once per application/binary (of which there could be multiple in a crate).

```Rust
#[entry(watchdog, psram(80))]
fn main() {
    // stuff
}
```

> I would prefer never to disable it

Not that I think this is bad to want or something, I'm just curious what you're worried about happening between the disable and re-enable? Or is this a matter of principal?

konsulten, 2024-06-21:
Im worried that I have made a bad OTA image and get stuck in it :/

MabezDev, 2024-09-16:
With the changes to how we initialize the hal, we can now make this a global configuration option. (Bear in mind this issue on the esp32 when implementing: https://github.com/esp-rs/esp-hal/issues/2105)

closed 2024-09-20 by SergioGasquez