esp-rs/esp-hal #1705: wifi example esp32

Arne-Berner, 2024-06-22:
When I try to build the [wifi example](https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/esp_wifi_access_point.rs) with the given [config.toml](https://github.com/esp-rs/esp-hal/blob/main/examples/.cargo/config.toml) and cargo.toml I get the following error:
/home/actualuser/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/bjoern/esp/esp-idf/components/wpa_supplicant/src/utils/common.c:141:(.text.wpa_get_ntp_timestamp+0x3d): undefined reference to `__bswapsi2'  
  collect2: error: ld returned 1 exit status                                                                                              
I tried installing the necessary drivers, but I couldn't get the example to work. 
I would assume it has something to do with needing /home/bjoern/... but I might be wrong. Any help is appreciated!

bjoernQ, 2024-06-24:
Do you run it from a clone of this repo or did you copy the code into your own project?

The example builds and runs fine for me on current `main` branch but I'm using toolchain 1.77

Arne-Berner, 2024-06-24:
I did copy the code. What could've gone wrong when copying the code 1 to 1 and using the config.toml? Is there a need to configure something else somewhere that is not visible in the example?

bjoernQ, 2024-06-24:
You are probably missing this: https://github.com/esp-rs/esp-hal/blob/9691141feddf7fff73fd5d734a99d213ae9966eb/examples/build.rs#L8-L10

You can also add it to `config.toml` instead of using `build.rs`. In that case you need to add `  "-C", "link-arg=-Trom_functions.x",`  to the rustflags

Arne-Berner, 2024-07-02:
After changing this line (and yes, I was missing that line) it still panics the same way.

bjoernQ, 2024-07-02:
> Do you think it should be changed to use the config.toml instead?

I agree that would be easier to discover but the examples cover a wide range of functionality and `esp-wifi` is an optional dependency. We unfortunately cannot have Rust-flags depending on features so we need to add the linker-arg via `build.rs`

Arne-Berner, 2024-07-02:
This makes sense. I guess doing more than write it in the README like you did is not possible. Thanks a lot for your help! I will just read it more carefully next time :)

closed 2024-07-02 by Arne-Berner