esp-rs/esp-hal #1706: H2: Enabling the IEEE802.15.4 peripheral causes the builtin JTAG peripheral to stop working

PortalRising, 2024-06-23:
I programmed the [ieee802154_receive_frame](https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/ieee802154_receive_frame.rs) example onto a [ESP32-H2-DEV-KIT-N4](https://www.waveshare.com/wiki/ESP32-H2-DEV-KIT-N4) by Waveshare using the builtin JTAG peripheral and the H2 becomes unresponsive and I am unable to be reprogram the board when the example is running.

# To Reproduce

Running this command in a local copy of the repository

    cargo esp32h2 --bin ieee802154_receive_frame 

Outputs this and just hangs until forcefully stopped

         Finished `release` profile [optimized + debuginfo] target(s) in 0.63s
         Running `espflash flash --monitor target/riscv32imac-unknown-none-elf/release/ieee802154_receive_frame`
    [2024-06-23T14:26:33Z INFO ] 🚀 A new version of espflash is available: v3.1.0
    [2024-06-23T14:26:33Z INFO ] Serial port: '/dev/ttyACM0'
    [2024-06-23T14:26:33Z INFO ] Connecting...

Finding a solution for #1623 may fix this issue if merged into the esp-ieee802154 crate as well.

bjoernQ, 2024-06-24:
> Finding a solution for https://github.com/esp-rs/esp-hal/issues/1623 may fix this issue if merged into the esp-ieee802154 crate as well.

Yes - I'm really interested in finding a solution for that. (You probably already know but to can get the board to be flashable again by manually putting it into download mode)

MabezDev, 2024-11-22:
Closing in favour of tracking in #1623

closed 2024-11-22 by MabezDev