esp-rs/esp-hal #1711: TIMG interrupt setup confusion

akriese, 2024-06-24:
Hi, I was playing around with the TIMG timers. My use case was to setup two periodic timers with interrupts. The only point where you can set up the timers with interrupt handlers is `TimerGroup::new()` which takes the `isr: Option<TimerInterrupts>` argument.
As I wanted both timers to be from the same group, I set the `TimerInterrupts` struct as follows:
```rust
let isr = TimerInterrupts {
  timer0_t0: Some(handler0),
  timer1_t0: Some(handler1),
  ..Default::default()
}
```
Thinking, that the `handler1` would be assigned to an alarm event from `timer1`.
I am not super proficient with some of the terminology, so I was confused as to why there are two interrupt handler registers per timer in a group (ESP32 manual calls them: `TIMGn_Tx_INT_T0_INT: An alarm event on timer 0 generates this interrupt.` and analogously for `timer0` in the same group). 
Looking the implementation up in the `esp-hal` source code, I saw that the `isr` handlers are assigned differently than expected:
```rust
            if let Some(handler) = isr.timer0_t0 {
                unsafe {
                    interrupt::bind_interrupt(Interrupt::TG0_T0_LEVEL, handler.handler());
                    interrupt::enable(Interrupt::TG0_T0_LEVEL, handler.priority()).unwrap();
                }
            }
```
and 
```rust 
                if let Some(handler) = isr.timer1_t0 {
                    unsafe {
                        interrupt::bind_interrupt(Interrupt::TG1_T0_LEVEL, handler.handler());
                        interrupt::enable(Interrupt::TG1_T0_LEVEL, handler.priority()).unwrap();
                    }
                }
```

I guess, `Interrupt::TG1_T0_LEVEL` stands for TIMG group 1, so I think there is some mistake here, as the interrupts for TIMG0 and TIMG1 should not be set from the `TimerGroup` implementation of one of the group instances.

Eventually, I adapted my solution to the implementation:
```rust
let isr = TimerInterrupts {
  timer0_t0: Some(handler0),
  timer0_t1: Some(handler1),
  ..Default::default()
}
```
... and it works, but it seems super unintuitive (to me) to have the handler for `timer1` be assigned to `TimerInterrupts.timer0_t1`.

I would appreciate if someone could shed some light on this or point to the part of documentation where these circumstances are explained.

Dominaezzz, 2024-06-24:
Yeah the names are little confusing.

I suppose this might've a little clearer?
```Rust
let isr = TimerInterrupts {
  group0_timer0: Some(handler0),
  group0_timer1: Some(handler1),
  ..Default::default()
}
```

Putting naming aside though, it also seems strange to me that the API allows you to initial the wrong timer group's interrupts.
```Rust
let isr = TimerInterrupts {
  group0_timer0: Some(handler0),
  group0_timer1: Some(handler1),
  ..Default::default()
}

let timg1 = TimerGroup::new(
    peripherals.TIMG1,
    &clocks,
    Some(isr),
);
```

The above code, I create a driver for timer group 1 but I initialised the isr for timer group 0.

The API should probably just look something like this.
```Rust
let isr = TimerInterrupts {
  timer0: Some(handler0),
  timer1: Some(handler1),
}

let timg1 = TimerGroup::new(
    peripherals.TIMG1,
    &clocks,
    Some(isr),
);
```
Then you can't specify the wrong group.

(Feel free to PR a name change if you like, though I'll probably do it this evening if no one else gets to it first)

akriese, 2024-06-24:
Thanks for the quick reply! The thing is, that the ESP32 manual states, that there are T0 and T1 interrupts for EACH timer in each group. What does that even mean? 
[image]

I agree with you that the API is probably not correct there. Your change to `TimerInterrupts` is what I would have imagined. But I wouldnt want to change anything there unless the above documentation weirdness can be ignored

Dominaezzz, 2024-06-24:
Ah yes I see what you're saying. If we expand `n` and `x` we get.
```
TIMG0_T0_INT_T1_INT
TIMG0_T0_INT_T0_INT
TIMG0_T1_INT_T1_INT
TIMG0_T1_INT_T0_INT
TIMG1_T0_INT_T1_INT
TIMG1_T0_INT_T0_INT
TIMG1_T1_INT_T1_INT
TIMG1_T1_INT_T0_INT
```

Which seems rather strange. I reckon it's a typo but I'll leave this to the espressif devs to clarify. At a glance I don't spot anything that justifies the two interrupts.

Even knowing this, T0 and T1, whatever their uses may be, they'll be handled by the same handler. So while there may be two interrupts per timer, there's only one handler per timer.

Dominaezzz, 2024-06-24:
esp-idf seems to agree with me about the typo https://github.com/espressif/esp-idf/blob/0479494e7abe5aef71393fba2e184b3a78ea488f/components/soc/esp32/include/soc/timer_group_reg.h#L556

So that doc should say
- TIMGn_WDT_INT: Generated when a watchdog timer interrupt stage times out.
- TIMGn_Tx_INT: An alarm event on timer x generates this interrupt.

bjoernQ, 2024-06-24:
> Putting naming aside though, it also seems strange to me that the API allows you to initialise the wrong timer group's interrupts.

Totally agree - if that would get cleaned up the naming should be as simple as `timer0`, `timer1` and `wdt`

PR #1715 (2024-06-25): TIMG: Fix interrupt handler setup
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] My  [... cut]
closed 2024-06-25 by jessebraham