esp-rs/esp-hal #1714: esp-storage hangs entire device on write

sikora77, 2024-06-24:
When I try to write to the FlashStorage the ESP32 hangs with no logs or panics regardless of write offset

bjoernQ, 2024-06-25:
Does the `flash_read_write` example (in-tree) work for you? Which exact dev-board/module/chip are you using?
You need to make sure to use the release profile (ESP32 really, really shouldn't access flash during reading/writing).

sikora77, 2024-06-25:
I am using an ESP32-WROOM-32, the example doesn't work, it hangs after the read, saying the flash size is 4194304 and the read section is just "ff" 32 times. I am using the release profile

bjoernQ, 2024-07-01:
What does `espflash board-info` print?

sikora77, 2024-07-02:
```
➜  ~ cargo espflash board-info
[2024-07-02T11:02:46Z INFO ] 🚀 A new version of cargo-espflash is available: v3.1.0
[2024-07-02T11:02:46Z INFO ] Serial port: '/dev/ttyUSB0'
[2024-07-02T11:02:46Z INFO ] Connecting...
[2024-07-02T11:02:47Z INFO ] Using flash stub
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       24:0a:c4:ee:02:e0

```

sikora77, 2024-09-04:
For some reason it stopped hanging, but writing to flash just doesn't work with the provided example code:
```
Flash size = 4194304

Read from 9000:  [68, 65, 6c, 6c, 6f, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff]
Written to 9000: [69, 67, 6f, 70, 70, 01, 02, 03, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff]
Read from 9000:  [68, 65, 6c, 60, 60, 01, 02, 03, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff]
Reset (CTRL-R in espflash) to re-read the persisted data.
```

sikora77, 2024-09-04:
after playing around with reading the flash with esptool it started working as intented

closed 2024-09-04 by sikora77

Flova, 2025-02-15:
I also encounter this issue. I have the same board and it sometimes randomly freezes on write.

sikora77, 2025-02-16:
what i noticed was wrong, i was importing the wrong way to write to flash, check your imports if they are the same as the example code for writing to flash