esp-rs/esp-hal #1728: Async Traits not Public

raphaelhetzel, 2024-06-27:
Hi,
I wanted to use `hal::spi::master::Spi` as an `embedded_hal_async::spi::SpiBus`.
Given the `asynch` module is not [public ](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/spi/master.rs#L1537) I don't see any way to do that.
Is this hidden by accident, or am I missing something?

bjoernQ, 2024-06-27:
What exactly sure what you are after but something like this works:

```rust
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    dma::*,
    dma_descriptors,
    gpio::Io,
    peripherals::Peripherals,
    prelude::*,
    spi::{
        master::{prelude::*, Spi},
        SpiMode,
    },
    system::SystemControl,
    timer::timg::TimerGroup,
};

#[main]
async fn main(_spawner: Spawner) {
    esp_println::println!("Init!");
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let timg0 = TimerGroup::new_async(peripherals.TIMG0, &clocks);
    esp_hal_embassy::init(&clocks, timg0);

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);
    let sclk = io.pins.gpio0;
    let miso = io.pins.gpio2;
    let mosi = io.pins.gpio4;
    let cs = io.pins.gpio5;

    let dma = Dma::new(peripherals.DMA);

    #[cfg(any(feature = "esp32", feature = "esp32s2"))]
    let dma_channel = dma.spi2channel;
    #[cfg(not(any(feature = "esp32", feature = "esp32s2")))]
    let dma_channel = dma.channel0;

    let (descriptors, rx_descriptors) = dma_descriptors!(32000);

    let spi = Spi::new(peripherals.SPI2, 100.kHz(), SpiMode::Mode0, &clocks)
        .with_pins(Some(sclk), Some(mosi), Some(miso), Some(cs))
        .with_dma(
            dma_channel.configure_for_async(false, DmaPriority::Priority0),
            descriptors,
            rx_descriptors,
        );

    do_spi(spi).await;
}

async fn do_spi<'a>(
    mut spi: esp_hal::spi::master::dma::SpiDma<
        'a,
        esp_hal::peripherals::SPI2,
        Channel0,
        esp_hal::spi::FullDuplexMode,
        esp_hal::Async,
    >,
) {
    esp_println::println!("here");
    let send_buffer = [0, 1, 2, 3, 4, 5, 6, 7];
    loop {
        let mut buffer = [0; 8];
        esp_println::println!("Sending bytes");
        embedded_hal_async::spi::SpiBus::transfer(&mut spi, &mut buffer, &send_buffer)
            .await
            .unwrap();
        esp_println::println!("Bytes received: {:?}", buffer);
        Timer::after(Duration::from_millis(5_000)).await;
    }
}
```

If you want to spawn an embassy task you would need to promote the lifetime to static (e.g. `make_static!`)

raphaelhetzel, 2024-07-01:
Thanks, I missed that.

Is there some example using SpiDMA as an embedded_hal_async SpiDevice (using embassy_embedded_hal or embedded_hal_bus)?

I am still trying to figure out why my code gets blocked at the flush at the end of a transaction (which seems to be waiting for some peripheral ref).

bjoernQ, 2024-07-01:
We used to have those examples but decided to consolidate the mostly redundant examples. `flush` just waits until the device isn't busy anymore - if you have some minimal-repro code for that issue we can have a look

raphaelhetzel, 2024-07-01:
I will try to fix it and will create a repro example if I'm sure it is an issue with the crate (most likely, I'm just using it wrong).
For now, it would be sufficient to understand who resets the register block in the flush busy method:
https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/spi/master.rs#L2943

bjoernQ, 2024-07-01:
`SPI_USR_COMMAND` is set by software to start the transfer and reset by hardware when done. 

You can read about the details in the TRM - e.g. for ESP32-C6: https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf#page=793

raphaelhetzel, 2024-07-04:
I further experimented with it, but the async spi stack in esp-hal (0.18) always gets stuck during a write.
Currently, the write gets stuck during `crate::dma::asynch::DmaTxFuture`.

I have attached a stripped-down version of my code as well as the sync equivalent that works.

Do you have any idea what might cause this?

Broken Async (embedded_hal_bus also did not work):
```rust
// SPDX-FileCopyrightText: © 2023 Technical University of Munich, Chair of Connected Mobility
// SPDX-License-Identifier: MIT
// Based on https://github.com/esp-rs/esp-hal/blob/main/esp32-hal/examples/embassy_hello_world.rs, https://github.com/esp-rs/esp-template/blob/main/src/main.rs & https://github.com/esp-rs/esp-wifi/blob/main/examples-esp32/examples/embassy_dhcp.rs

#![no_std]
#![no_main]

extern crate alloc;
pub mod wifi;
pub mod epaper_display_impl;
use embedded_hal_async::{delay::DelayNs, spi::SpiDevice};
use esp_backtrace as _;
use hal::prelude::*;

pub mod epaper_v2;

use hal::spi::master::prelude::*;

#[global_allocator]
static ALLOCATOR: esp_alloc::EspHeap = esp_alloc::EspHeap::empty();

static RNG: once_cell::sync::OnceCell<hal::rng::Rng> = once_cell::sync::OnceCell::new();

const ESP_GETRANDOM_ERROR: u32 = getrandom::Error::CUSTOM_START + 1;

const NODE_ID: uuid::Uuid = uuid::uuid!("0827240a-3050-4604-bf3e-564c41c77106");

fn init_heap() {
    const HEAP_SIZE: usize = 32 * 1024;
    static mut HEAP: core::mem::MaybeUninit<[u8; HEAP_SIZE]> = core::mem::MaybeUninit::uninit();

    unsafe {
        ALLOCATOR.init(HEAP.as_mut_ptr() as *mut u8, HEAP_SIZE);
    }
}

#[entry]
fn main() -> ! {
    esp_println::logger::init_logger(log::LevelFilter::Info);
    esp_println::println!("Start Edgeless Embedded.");

    // https://github.com/esp-rs/esp-template/blob/main/src/main.rs
    init_heap();

    let peripherals = hal::peripherals::Peripherals::take();
    #[allow(unused_variables)]
    let io = hal::gpio::Io::new(peripherals.GPIO, peripherals.IO_MUX);
    let system = hal::system::SystemControl::new(peripherals.SYSTEM);

    let clocks = hal::clock::ClockControl::max(system.clock_control).freeze();
    let timer_group0 = hal::timer::timg::TimerGroup::new_async(peripherals.TIMG0, &clocks);
    let timer_group1 = hal::timer::timg::TimerGroup::new(peripherals.TIMG1, &clocks, None);

    esp_hal_embassy::init(&clocks, timer_group0);

    let dma = hal::dma::Dma::new(peripherals.DMA);
    let dma_channel = dma.spi2channel;
    let (mut descriptors, mut rx_descriptors) = hal::dma_descriptors!(32000);

    static DESCRIPTORS : static_cell::StaticCell<[hal::dma::DmaDescriptor; 8]> = static_cell::StaticCell::new();
    let desc = DESCRIPTORS.init_with(|| descriptors);
[... 247 lines cut ...]
    log::info!("Reset");
    rst.set_high();
    delay.delay_ms(10).await;
    rst.set_low();
    delay.delay_ms(10).await;
    rst.set_high();
    delay.delay_ms(200).await;

    log::info!("Wait Idle");
    while busy.is_high() {
        delay.delay_ms(10).await;
    }

    log::info!("SPI DC");

    dc.set_low();

    log::info!("SPI Write");

    log::info!("{:?}", spi.write(&[18]));

    log::info!("SPI_WRITE_DONE");
}

```

bjoernQ, 2024-07-05:
Ok took a while to get your example to compile.

It works for me on ESP32-C3 but I can confirm I see it waiting forever on ESP32

raphaelhetzel, 2024-07-05:
Sorry for that, is there anything I can do to simplify it for repro?

bjoernQ, 2024-07-05:
No problem!

Here is a minimal repro (just change the `embassy_spi.rs` example to look like this)
```rust
//! Embassy SPI
//!
//! Folowing pins are used:
//! SCLK    GPIO0
//! MISO    GPIO2
//! MOSI    GPIO4
//! CS      GPIO5
//!
//! Depending on your target and the board you are using you have to change the
//! pins.
//!
//! Connect MISO and MOSI pins to see the outgoing data is read as incoming
//! data.
//!
//! This is an example of running the embassy executor with SPI.

//% CHIPS: esp32 esp32c2 esp32c3 esp32c6 esp32h2 esp32s2 esp32s3
//% FEATURES: async embassy embassy-time-timg0 embassy-generic-timers

#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    dma::*,
    dma_descriptors,
    gpio::Io,
    peripherals::Peripherals,
    prelude::*,
    spi::{
        master::{prelude::*, Spi},
        SpiMode,
    },
    system::SystemControl,
    timer::timg::TimerGroup,
};

#[main]
async fn main(_spawner: Spawner) {
    esp_println::println!("Init!");
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let timg0 = TimerGroup::new_async(peripherals.TIMG0, &clocks);
    esp_hal_embassy::init(&clocks, timg0);

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);
    let sclk = io.pins.gpio0;
    let miso = io.pins.gpio2;
    let mosi = io.pins.gpio4;
    let cs = io.pins.gpio5;

    let dma = Dma::new(peripherals.DMA);

    #[cfg(any(feature = "esp32", feature = "esp32s2"))]
    let dma_channel = dma.spi2channel;
    #[cfg(not(any(feature = "esp32", feature = "esp32s2")))]
    let dma_channel = dma.channel0;

    let (descriptors, rx_descriptors) = dma_descriptors!(32000);

    let mut spi = Spi::new(peripherals.SPI2, 100.kHz(), SpiMode::Mode0, &clocks)
        .with_pins(Some(sclk), Some(mosi), Some(miso), Some(cs))
        .with_dma(
            dma_channel.configure_for_async(false, DmaPriority::Priority0),
            descriptors,
            rx_descriptors,
        );

    let send_buffer = [0; 5];
    loop {
        esp_println::println!("Sending bytes");
        embedded_hal_async::spi::SpiBus::write(&mut spi, &send_buffer)
            .await
            .unwrap();
        esp_println::println!("Done");
        Timer::after(Duration::from_millis(1_000)).await;
    }
}
```

Very small writes (< 17 in my tests) will make ESP32 never finish awaiting the write. It works on e.g. ESP32-C3

PR #1760 (2024-07-15): Fix reading/writing small buffers
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] My  [... cut]
closed 2024-07-15 by SergioGasquez