esp-rs/esp-hal #1747: Reduce / eliminate cache misses in interrupt handling

bjoernQ, 2024-07-04:
Follow up on #1162 - there are some more insights in the linked issue

Remaining tasks

- [ ] Replace `const`s this `static`'s if they're referenced in interrupt handling code
- [ ] Put jump tables in RAM
    - Linker magic to place them in RAM on a best effort basis

There is even more to put into RAM to avoid hitting flash. To really see where code accesses flash its best to look at disassembled code

We _need_ to make this opt-in since it will decrease remaining useable RAM which might be a problem for some applications.

Given we don't want to introduce more features this means need #1111 first

MabezDev, 2024-09-26:
Now with some configuration options we can opt into placing more of these things into IRAM

bugadani, 2024-09-26:
If we aim to reduce cache misses, we might want to optimize IRAM code to size. The attribute [exists](https://github.com/rust-lang/rust/issues/54882#issuecomment-457824508) in nightly, so if we want a config to enable nightly-specific features we can do this automagically for all IRAM code.

PR #2331 (2024-10-18): Add option to place more `.rodata` in RAM for performance
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_prop [... cut]
closed 2024-10-18 by MabezDev