esp-rs/esp-hal #1766: examples/esp_wifi_access_point: Stations disassociate frequently due to beacon loss.

frederictobiasc, 2024-07-06:
Hi!

I'm trying to get a project with an ESP32-C3 as Wi-Fi AP running. However, even the examples lead to the following error:

When I associate a station with the network of the esp32-c3, the station goes in a reconnection loop. After a few seconds, the driver signals **missed beacons** and finally disassociates.
Then, it reconnects, and the loop starts again.

 I also tried different esp32c3's as well as different Wi-Fi stations, to make sure it's not a hardware fault.

Logs:
- [dmesg.log](https://github.com/user-attachments/files/16115750/dmesg.log)
- [ping.log](https://github.com/user-attachments/files/16115751/ping.log)

## How to reproduce

1. Clone this repo (esp-hal)
2. run `cargo xtask run-example esp-hal esp32c3 esp_wifi_access_point`
3. associate a wifi station to the network "esp-wifi"
4. Observe the dmesg or check the ping log

bjoernQ, 2024-07-15:
I just tried this and it works fine for me. One of my ESP32-C3 dev-kits frequently reset because of BROWNOUT_RST but another worked fine.

What I did
- run `cargo xtask run-example examples esp32c3 wifi_access_point` from the repo (commit 0363169084ac6c25ba40196a977f8cd789652880)
- use my Android phone, connect to `esp-wifi`AP and get a shell
- run `ping`

```
❯ .\adb.exe shell
SDW:/ $ ping 192.168.2.1
PING 192.168.2.1 (192.168.2.1) 56(84) bytes of data.
64 bytes from 192.168.2.1: icmp_seq=1 ttl=64 time=8.34 ms
64 bytes from 192.168.2.1: icmp_seq=2 ttl=64 time=52.8 ms
64 bytes from 192.168.2.1: icmp_seq=3 ttl=64 time=2.91 ms
64 bytes from 192.168.2.1: icmp_seq=4 ttl=64 time=23.6 ms
64 bytes from 192.168.2.1: icmp_seq=5 ttl=64 time=2.85 ms
64 bytes from 192.168.2.1: icmp_seq=6 ttl=64 time=3.23 ms
64 bytes from 192.168.2.1: icmp_seq=7 ttl=64 time=26.5 ms
64 bytes from 192.168.2.1: icmp_seq=8 ttl=64 time=3.28 ms
64 bytes from 192.168.2.1: icmp_seq=9 ttl=64 time=4.32 ms
64 bytes from 192.168.2.1: icmp_seq=10 ttl=64 time=31.6 ms
64 bytes from 192.168.2.1: icmp_seq=11 ttl=64 time=2.57 ms
64 bytes from 192.168.2.1: icmp_seq=12 ttl=64 time=6.97 ms
64 bytes from 192.168.2.1: icmp_seq=13 ttl=64 time=2.83 ms
64 bytes from 192.168.2.1: icmp_seq=14 ttl=64 time=71.9 ms
64 bytes from 192.168.2.1: icmp_seq=15 ttl=64 time=2.31 ms
64 bytes from 192.168.2.1: icmp_seq=16 ttl=64 time=3.24 ms
64 bytes from 192.168.2.1: icmp_seq=17 ttl=64 time=4.30 ms
64 bytes from 192.168.2.1: icmp_seq=18 ttl=64 time=5.87 ms
64 bytes from 192.168.2.1: icmp_seq=19 ttl=64 time=5.12 ms
64 bytes from 192.168.2.1: icmp_seq=20 ttl=64 time=2.36 ms
64 bytes from 192.168.2.1: icmp_seq=21 ttl=64 time=15.3 ms
64 bytes from 192.168.2.1: icmp_seq=22 ttl=64 time=95.8 ms
64 bytes from 192.168.2.1: icmp_seq=23 ttl=64 time=3.09 ms
64 bytes from 192.168.2.1: icmp_seq=24 ttl=64 time=4.75 ms
64 bytes from 192.168.2.1: icmp_seq=25 ttl=64 time=22.6 ms
64 bytes from 192.168.2.1: icmp_seq=26 ttl=64 time=18.3 ms
64 bytes from 192.168.2.1: icmp_seq=27 ttl=64 time=17.1 ms
64 bytes from 192.168.2.1: icmp_seq=28 ttl=64 time=17.1 ms
64 bytes from 192.168.2.1: icmp_seq=29 ttl=64 time=4.60 ms
64 bytes from 192.168.2.1: icmp_seq=30 ttl=64 time=7.98 ms
64 bytes from 192.168.2.1: icmp_seq=31 ttl=64 time=5.45 ms
64 bytes from 192.168.2.1: icmp_seq=32 ttl=64 time=3.68 ms
64 bytes from 192.168.2.1: icmp_seq=33 ttl=64 time=108 ms
64 bytes from 192.168.2.1: icmp_seq=34 ttl=64 time=59.2 ms
64 bytes from 192.168.2.1: icmp_seq=35 ttl=64 time=15.4 ms
64 bytes from 192.168.2.1: icmp_seq=36 ttl=64 time=12.9 ms
64 bytes from 192.168.2.1: icmp_seq=37 ttl=64 time=3.28 ms
64 bytes from 192.168.2.1: icmp_seq=38 ttl=64 time=4.97 ms
64 bytes from 192.168.2.1: icmp_seq=39 ttl=64 time=10.4 ms
64 bytes from 192.168.2.1: icmp_seq=40 ttl=64 time=21.6 ms
64 bytes from 192.168.2.1: icmp_seq=41 ttl=64 time=6.49 ms
64 bytes from 192.168.2.1: icmp_seq=42 ttl=64 time=2.67 ms
64 bytes from 192.168.2.1: icmp_seq=43 ttl=64 time=17.4 ms
64 bytes from 192.168.2.1: icmp_seq=44 ttl=64 time=103 ms
64 bytes from 192.168.2.1: icmp_seq=45 ttl=64 time=40.6 ms
64 bytes from 192.168.2.1: icmp_seq=46 ttl=64 time=14.8 ms
64 bytes from 192.168.2.1: icmp_seq=47 ttl=64 time=17.7 ms
64 bytes from 192.168.2.1: icmp_seq=48 ttl=64 time=16.8 ms
64 bytes from 192.168.2.1: icmp_seq=49 ttl=64 time=3.94 ms
64 bytes from 192.168.2.1: icmp_seq=50 ttl=64 time=2.92 ms
64 bytes from 192.168.2.1: icmp_seq=51 ttl=64 time=28.1 ms
64 bytes from 192.168.2.1: icmp_seq=52 ttl=64 time=72.2 ms
64 bytes from 192.168.2.1: icmp_seq=53 ttl=64 time=2.98 ms
64 bytes from 192.168.2.1: icmp_seq=54 ttl=64 time=49.5 ms
64 bytes from 192.168.2.1: icmp_seq=55 ttl=64 time=3.33 ms
64 bytes from 192.168.2.1: icmp_seq=56 ttl=64 time=3.38 ms
64 bytes from 192.168.2.1: icmp_seq=57 ttl=64 time=4.26 ms
64 bytes from 192.168.2.1: icmp_seq=58 ttl=64 time=12.2 ms
64 bytes from 192.168.2.1: icmp_seq=59 ttl=64 time=3.31 ms
64 bytes from 192.168.2.1: icmp_seq=60 ttl=64 time=48.7 ms
64 bytes from 192.168.2.1: icmp_seq=62 ttl=64 time=49.9 ms
64 bytes from 192.168.2.1: icmp_seq=63 ttl=64 time=34.3 ms
64 bytes from 192.168.2.1: icmp_seq=64 ttl=64 time=119 ms
64 bytes from 192.168.2.1: icmp_seq=65 ttl=64 time=3.47 ms
64 bytes from 192.168.2.1: icmp_seq=66 ttl=64 time=51.4 ms
64 bytes from 192.168.2.1: icmp_seq=67 ttl=64 time=17.4 ms
64 bytes from 192.168.2.1: icmp_seq=68 ttl=64 time=15.6 ms
64 bytes from 192.168.2.1: icmp_seq=69 ttl=64 time=15.6 ms
64 bytes from 192.168.2.1: icmp_seq=70 ttl=64 time=4.45 ms
64 bytes from 192.168.2.1: icmp_seq=71 ttl=64 time=2.91 ms
64 bytes from 192.168.2.1: icmp_seq=72 ttl=64 time=2.62 ms
64 bytes from 192.168.2.1: icmp_seq=73 ttl=64 time=35.7 ms
64 bytes from 192.168.2.1: icmp_seq=74 ttl=64 time=75.4 ms
64 bytes from 192.168.2.1: icmp_seq=75 ttl=64 time=15.2 ms
64 bytes from 192.168.2.1: icmp_seq=76 ttl=64 time=13.4 ms
64 bytes from 192.168.2.1: icmp_seq=77 ttl=64 time=34.5 ms
64 bytes from 192.168.2.1: icmp_seq=78 ttl=64 time=4.32 ms
64 bytes from 192.168.2.1: icmp_seq=79 ttl=64 time=3.22 ms
64 bytes from 192.168.2.1: icmp_seq=80 ttl=64 time=75.8 ms
64 bytes from 192.168.2.1: icmp_seq=81 ttl=64 time=3.42 ms
64 bytes from 192.168.2.1: icmp_seq=82 ttl=64 time=4.33 ms
64 bytes from 192.168.2.1: icmp_seq=83 ttl=64 time=89.6 ms
64 bytes from 192.168.2.1: icmp_seq=84 ttl=64 time=3.98 ms
```

Maybe you see frequent resets on the ESP32-C3 like I do for one of my boards?

Closing for now - feel free to reopen with updated information

frederictobiasc, 2024-07-16:
Hi @bjoernQ and thanks for checking.

I have four ESP32-C3-DevKitC-02 v1.1 at hand and tested all of them. Two of them were never used before. I made sure to use the exact same commit as you did (https://github.com/esp-rs/esp-hal/commit/0363169084ac6c25ba40196a977f8cd789652880).

I used the following stations for testing:

- Qualcomm QCNFA765
- Intel Wireless 8265
- Intel Wireless 9560
- Android Phone (Snapdragon 835)

I confirm that the Android Phone seems to be able to establish a reliable connection. I tested this with ADB shell like you did. All other devices disassociated because of **Beacon Loss** and show a similar ping report like the one in my report above. There is no improvement when I provide the devkits with 3.3v directly from my lab PSU.

Using [ESP-IDFs softAP example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/softAP) works well on all my four devkits.

Could you please check with another station apart from your android phone?

bjoernQ, 2024-07-17:
I tried again with my Windows box.

```
❯ netsh wlan show  interfaces

Es ist 1 Schnittstelle auf dem System vorhanden:

    Name                   : WLAN
    Beschreibung            : Killer(R) Wi-Fi 6 AX1650i 160MHz Wireless Network Adapter (201NGW)
    GUID                   : be67aad6-52c7-45b8-94a9-b2d0338e53f0
    Physische Adresse       : 40:1c:83:10:a4:2d
    Benutzeroberflächentyp         : Primär
    Status                  : Verbunden
    SSID                   : esp-wifi
    BSSID                  : 36:b4:72:4c:44:18
    Netzwerktyp            : Infrastruktur
    Funktyp                   : 802.11n
    Authentifizierung   : Offen
    Verschlüsselung                 : Keine
    Verbindungsmodus        : Profil
    Band                   : 2,4 GHz
    Kanal                : 1
    Empfangsrate (MBit/s)  : 6
    Übertragungsrate (MBit/s) : 6
    Signal              : 99%
    Profil                 : esp-wifi
```

The ping result looks a bit worse compared to my Android phone
```
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=463ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=532ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.100: Zielhost nicht erreichbar.
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=370ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=438ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=366ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
Antwort von 192.168.2.100: Zielhost nicht erreichbar.
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=34ms TTL=64
Zeitüberschreitung der Anforderung.
Antwort von 192.168.2.1: Bytes=32 Zeit=696ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=75ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=867ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=930ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Zeitüberschreitung der Anforderung.
Antwort von 192.168.2.1: Bytes=32 Zeit=397ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=6ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=472ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=734ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
Zeitüberschreitung der Anforderung.
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=835ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=12ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=966ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
```

There are especially some bad peaks

Maybe worth to try tweaking some settings: https://github.com/esp-rs/esp-hal/blob/main/esp-wifi/tuning.md

(Sorry the output is from a German Windows installation but I guess it's not too hard to guess what the messages say)

frederictobiasc, 2024-07-18:
Hi @bjoernQ,

I investigated the beacon error and there seems to be something seriously wrong with esp32-c3's Wi-Fi setup via `esp-hal`.

I did two wireshark captures on the same devkit hardware with two different firmwares flashed:

## Capture of [ESP-IDFs softAP example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/softAP)

[idf_softAP.zip](https://github.com/user-attachments/files/16286440/idf_softAP.zip)

Flash command: `idf.py -p /dev/ttyUSB0 flash monitor`

## Capture of [`esp-hal/examples/wifi_access_point`](https://github.com/esp-rs/esp-hal/blob/0363169084ac6c25ba40196a977f8cd789652880/examples/src/bin/wifi_access_point.rs)

[esp-rs.zip](https://github.com/user-attachments/files/16286441/esp-rs.zip)

Flash command: `cargo xtask run-example esp-hal esp32c3 esp_wifi_access_point`

## Observations

The MAC address used as transmitter/source address by the ESP differs:
|example|MAC|
|-|-|
|ESP-IDF softAP|7c:df:a1:af:4f:75|
|esp-rs wifi\_access\_point|7e:df:a1:af:4f:74|
|reported by espflash & esptool.py|7c:df:a1:af:4f:74|

The softAP example works flawlessly with every station I tried. The capture shows beacon frames at a 10 per second rate. This corresponds to the typical interval of 100 [TU](https://en.wikipedia.org/wiki/TU_(time_unit))'s.

On the other hand, the capture of `esp-rs` `wifi_access_point` example shows that only a single beacon frame is generated at the startup. This leads to the disassociation, because loss of beacons is interpreted as a signal loss.

Maybe there is some timer/interrupt misconfigured?

## Info: Capture Wi-Fi Beacon Frames

To capture beacon frames, the Wi-Fi adapter needs to be put into monitoring mode. I did this on my Arch notebook like follows:

Replace wlp3s0 with your interface's name.

```
systemctl stop NetworkManager
iwconfig wlp3s0 mode monitor
ip link set wlp3s0 up
```

Then start Wireshark and start capturing on `wlp3s0`.

frederictobiasc, 2024-07-18:
Launching the example with feature `wifi-logs` enabled leads to: the endlessly repeating message

`INFO - The previous beacon was not Tx done`

```
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 15, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "DE", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 10, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
INFO - pp rom version: 9387209

INFO - net80211 rom version: 9387209

INFO - I (301) wifi:
INFO - wifi driver task: 00, prio:253, stack:6656, core=0
INFO - 

INFO - I (312) wifi:
INFO - wifi firmware version: e309346
INFO - 

INFO - I (318) wifi:
INFO - wifi certification version: v7.0
INFO - 

INFO - I (327) wifi:
INFO - config NVS flash: disabled
INFO - 

INFO - I (334) wifi:
INFO - config nano formating: disabled
INFO - 

INFO - I (343) wifi:
INFO - Init data frame dynamic rx buffer num: 32
INFO - 

INFO - I (352) wifi:
INFO - Init static rx mgmt buffer num: 10
INFO - 

INFO - I (361) wifi:
INFO - Init management short buffer num: 32
INFO - 

INFO - I (369) wifi:
INFO - Init dynamic tx buffer num: 32
INFO - 

INFO - I (378) wifi:
INFO - Init static tx FG buffer num: 2
INFO - 

INFO - I (386) wifi:
INFO - Init static rx buffer size: 2212
INFO - 

INFO - I (394) wifi:
INFO - Init static rx buffer num: 10
INFO - 

INFO - I (402) wifi:
INFO - Init dynamic rx buffer num: 32
INFO - 

INFO - V (411) wifi:
INFO - hint=255, act_dur=<0,120>, pas_dur=360, nchans=14 history_num=0
INFO - 

INFO - W (424) wifi:
INFO - Affected by the ESP-NOW encrypt num, set the max connection num to 10
INFO - 

INFO - W (435) wifi:
INFO - Affected by the ESP-NOW encrypt num, set the max connection num to 10
INFO - 

wifi_set_configuration returned Ok(())
INFO - V (483) wifi:
INFO - ht40u freq=2422, chan=1
INFO - 

INFO - D (485) wifi:
INFO - filter: set rx policy=0
INFO - 

INFO - I (489) wifi:
INFO - mode : softAP (7e:df:a1:af:4f:74)
INFO - 

INFO - D (496) wifi:
INFO - filter: set rx policy=9
INFO - 

INFO - V (504) wifi:
INFO - ht40u freq=2422, chan=1
INFO - 

INFO - I (512) wifi:
INFO - Total power save buffer number: 16
INFO - 

INFO - I (520) wifi:
INFO - Init max length of beacon: 752/752
INFO - 

INFO - I (529) wifi:
INFO - Init max length of beacon: 752/752
INFO - 

INFO - I (539) wifi:
INFO - Set ps type: 0, coexist: 0

INFO - 

INFO - I (546) wifi:
INFO - set country: cc=DE schan=1 nchan=13 policy=1

INFO - 

is wifi started: Ok(true)
Ok(EnumSet(AccessPoint))
INFO - Set hardware address: Ethernet(Address([126, 223, 161, 175, 79, 116]))
Start busy loop on main. Connect to the AP `esp-wifi` and point your browser to http://192.168.2.1:8080/
Use a static IP in the range 192.168.2.2 .. 192.168.2.255, use gateway 192.168.2.1
INFO - W (743) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (846) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (948) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1051) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1153) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1255) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1358) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1460) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1563) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1665) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1767) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1870) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (1972) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2075) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2177) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2280) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2381) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2483) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2586) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2688) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2791) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2893) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (2996) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (3098) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (3200) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (3303) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (3405) wifi:
INFO - The previous beacon was not Tx done
INFO - 

INFO - W (3508) wifi:
INFO - The previous beacon was not Tx done
INFO - 
```

bjoernQ, 2024-07-19:
`The previous beacon was not Tx done` is an interesting observation. Reopening the issue

bjoernQ, 2024-07-24:
Ok that's interesting. I see the same logs for ESP32-C3 but not e.g. on ESP32-C6. 

<details>
<summary>There I also see quite nice pings from my Windows box</summary>
```
❯ ping -t 192.168.2.1

Ping wird ausgeführt für 192.168.2.1 mit 32 Bytes Daten:
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=5ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=11ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=14ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=6ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=6ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=25ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=5ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=12ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=3ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=4ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=2ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit=1ms TTL=64
Antwort von 192.168.2.1: Bytes=32 Zeit<1ms TTL=64
```
</details>

frederictobiasc, 2024-09-06:
Hi @bjoernQ. I just tried this with the v0.20.1 release as well as latest master. The issue is still there, unfortunately.
I also ran into a new issue: https://github.com/esp-rs/esp-hal/issues/2111

Is there anything I can do to help to get this fixed?

bjoernQ, 2024-09-09:
I think we need to update the wifi drivers. When running the esp-idf softAp example using the esp-idf commit corresponding to our version of the drivers I continously get this

```
D (275601) wifi:no bcast bss, send deauth: reason not authed(6)
D (295571) wifi:no bcast bss, send deauth: reason not authed(6)
D (315571) wifi:no bcast bss, send deauth: reason not authed(6)
D (335581) wifi:no bcast bss, send deauth: reason not authed(6)
D (355581) wifi:no bcast bss, send deauth: reason not authed(6)
D (375591) wifi:no bcast bss, send deauth: reason not authed(6)
D (395601) wifi:no bcast bss, send deauth: reason not authed(6)
D (415611) wifi:no bcast bss, send deauth: reason not authed(6)
D (435611) wifi:no bcast bss, send deauth: reason not authed(6)
D (455621) wifi:no bcast bss, send deauth: reason not authed(6)
```

Not the same error but doesn't look too good to me

closed 2024-09-30 by JurajSadel