esp-rs/esp-hal #1768: Feature Request: iterator as input for RMT

tschundler, 2024-07-07:
Right now, the `smart_leds` support is limited significantly, needing to pre-allocate a buffer for RMT symbols. Building the RMT symbols is already done internally as an iterator, and read as an iterator. There is time for ~2k instructions per byte transmitted, which should be plenty for translating symbols and keeping the RMT's ring buffer full.

Does that seem like a reasonable feature? `transmit_iter`? I'll likely try it out myself in the next week if I have time. Also I'm curious to see if I can drive multiple strips at once that way with embassy.

Does anyone have opinions about when to refill the buffer? eg I wonder if I might get better results refilling in 16 byte chunks instead of one it is half depleted. That way in async if something eats a little extra time, there's more headroom. Does that seem reasonable?

related: #1749, #1710
my last experimenting with older version of the RMT driver: #355

tschundler, 2024-07-09:
Playing a little, I have it working in blocking mode. I can generate some patterns with no buffers at all.

I did need to use a different signature. When iterating an array, `Iterator<Item = &'a T> + 'a` works, but when generating content on the fly, what is the reference to? It's gone when `.next()` finishes. So instead I need this signature for `transmit`:

```rust
    fn transmit<D, R, T: Into<u32> + Copy>(self, data: D) -> SingleShotTxTransaction<Self, R, T>
    where
        Self: Sized,
        D: IntoIterator<Item = T, IntoIter = R>,
        R: Iterator<Item = T>,

```

which is not backwards-compatible, since it can't take `&[u32;n]` as the data. So should I have `transmit` + `transmit_iter` (or better name?) where `transmit` calls `transmit_iter(data.iter().cloned())`

tschundler, 2024-07-10:
An important question as I continue to work on this:

How important is API stability right now? I can change the `transmit` function or add `transmit_iter`, which means nothing breaks. I've added an underflow error to help diagnose misbehavior - should it be part of a separate error enum used by `transmit_iter`?

And for the SmartLEDsAdapter, change it, or add `SmartLEDsUnbuffered`?

tschundler, 2024-07-14:
I should update in my own thread...

 - [x] Iterator input for RMT
 - [x] Iterator from SmartLEDs for unbuffered SmartLEDs
 - [ ] Async RMT iterator works only with no interrupts and very little else going on #1779
 - [ ] Async SmartLEDs #1710 blocked on #1779
 
In the coming days I'll probably submit a pull requests related to RMT from iterator & SmartLEDs HAL via iterator. I am curious about any input before I submit a PR.

Dominaezzz, 2024-07-24:
> How important is API stability right now? I can change the `transmit` function or add `transmit_iter`, which means nothing breaks. I've added an underflow error to help diagnose misbehavior - should it be part of a separate error enum used by `transmit_iter`?

Feel free to break the API but be sure to provide a migration guide in the PR comment.