esp-rs/esp-hal #1774: Create a Migration Guide document

SergioGasquez, 2024-07-09:
This idea was discussed in the past, having a Migration Guide document would help to improve the migration guide as we can ask the author of each contribution that requires a paragraph on it to write it.

If we end up adding such document, it might be worth to add a sentence in the PR template (having a CI check sounds a little too much imho)

MabezDev, 2024-07-09:
We have the start of these in our release notes, I'm wondering if a separate document is better than just using those :thinking:

SergioGasquez, 2024-07-09:
We can definitely squeeze them into the changelog, that's totally fine with me. The important part, for me, is to allow the author of a change to write them instead of one person doing them all at release time.

trinhanhtuan, 2024-08-21:
There is a fact that the document for essp-hal is a bit poor and not keep up with the releases.
I can join the team to contribute to this guide or making a  better living esp-hal documentation.
How should I start?

MabezDev, 2024-08-21:
@trinhanhtuan documentation is separate from a migration guide. PRs to improve docs can be filed at anytime and would be most welcome :).

closed 2024-09-02 by MabezDev