esp-rs/esp-hal #1781: Running `embassy_hello_world` example stand-alone

lure23, 2024-07-10:
I can build & run the example code using `cargo xtask run-example`, but when moving away from the `xtask` system, I haven't been able to compile a single example.

I'd like to be able to have the `embassy_hello_world` sample work:

- without `cargo xtask`
- with pulling dependencies from `crates.io`

>I am aware that `esp-template` is the recommended way to get my own sails up. However, it restricts to targeting a single ESP32 variant, and I'd like to target a few, at once. Since `cargo xtask` is "just" a wrapper on `cargo build` and `espflash`, this  should be possible. Right? :)

## `cargo xtask` (works)

```
$ cd esp-hal
```

```
$ cargo xtask run-example examples esp32c3 embassy_hello_world
[...]
Init!
Bing!
Hello world from embassy using esp-hal-async!
````

## `cargo build` with file system dependencies

I should be able to replicate the above simply by `cargo` commands, in the `examples` folder. Picking `features` from the output of the previous command:

```
[2024-07-10T13:59:42Z INFO  xtask]   Features: embassy,esp-hal-embassy/integrated-timers
```

```
$ cd examples
```

```
$ cargo build --release --bin embassy_hello_world --features=esp32c3,embassy,esp-hal-embassy/integrated-timers
```

This eventually fails:

```
   Compiling basic-toml v0.1.9
error: cfg(portable_atomic_unsafe_assume_single_core) does not compatible with this target;
       if you need cfg(portable_atomic_unsafe_assume_single_core) support for this target,
       please submit an issue at <https://github.com/taiki-e/portable-atomic>
   --> /home/ubuntu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.6.0/src/lib.rs:362:1
    |
362 | / compile_error!(
363 | |     "cfg(portable_atomic_unsafe_assume_single_core) does not compatible with this target;\n\
364 | |      if you need cfg(portable_atomic_unsafe_assume_single_core) support for this target,\n\
365 | |      please submit an issue at <https://github.com/taiki-e/portable-atomic>"
366 | | );
    | |_^

error[E0432]: unresolved import `super::core_atomic`
  --> /home/ubuntu/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.6.0/src/imp/x86.rs:19:12
   |
19 | use super::core_atomic::{
   |            ^^^^^^^^^^^ could not find `core_atomic` in `super`
   |
[...]
```

Q: What would be the right build (or run) command, to replicate the success of `cargo xtask`?

Once I get this to work, I'll continue with pointing the dependencies to `crates.io`.

jessebraham, 2024-07-11:
Thank you for opening this issue. We are aware that this currently quite difficult, sorry that you have been having problems. We're planning on publishing a new round of packages next week, and following that our next release cycle will be focused on improving usability. I will be sure to mention this issue during our team's planning meeting next week.

In the meantime, while this is not a true solution, I have made a template for using `esp-hal` with `embassy`, which you can find here:  
https://github.com/jessebraham/esp-hal-template

It's still quite basic, but at least should generate the boilerplate needed to build an Embassy application without relying on our `xtask` package. If you have any problems or additional requests, please feel free to open one or more issues on that repository. I threw this together pretty quick yesterday, so there may be some minor issues; I plan to expand on it a bit this weekend.

lure23, 2024-07-12:
No worries! The repo you linked to was helpful in seeing which dependencies are really needed. 

Glad to be on this ride!  Nice to know momentum is there.

lure23, 2024-07-17:
Looks fine now, with `esp-hal` 0.19.0 et.al.

closed 2024-07-17 by lure23