esp-rs/esp-hal #1792: Missing ErasedTimer from imports

skrapi, 2024-07-14:
Hi,

I am trying to use esp-hal for the esp32c3 on the ESP Rustboard. I generated the project from 

```
cargo generate esp-rs/esp-template
```

and then copied the text from the embassy_hello_world.rs example. I added the missing imports

```
[package]
name = "embedded-controller"
version = "0.1.0"
authors = ["Sylvan <sylvan@hey.com>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
embassy-executor = "0.5.0"
embassy-time = "0.3.1"
esp-backtrace = { version = "0.12.0", features = [
    "esp32c3",
    "exception-handler",
    "panic-handler",
    "println",
] }
esp-hal = { version = "0.18.0", features = ["esp32c3", "async"] }
esp-hal-embassy = { version = "0.1.0", features = ["esp32c3", "time-timg0"] }
esp-println = { version = "0.9.1", features = ["esp32c3", "log"] }
log = { version = "0.4.21" }
static_cell = "2.1.0"

[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1        # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false
```

src/main.rs

```
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    peripherals::Peripherals,
    prelude::*,
    system::SystemControl,
    timer::{timg::TimerGroup, ErasedTimer, OneShotTimer},
};

// When you are okay with using a nightly compiler it's better to use https://docs.rs/static_cell/2.1.0/static_cell/macro.make_static.html
macro_rules! mk_static {
    ($t:ty,$val:expr) => {{
        static STATIC_CELL: static_cell::StaticCell<$t> = static_cell::StaticCell::new();
        #[deny(unused_attributes)]
        let x = STATIC_CELL.uninit().write(($val));
        x
    }};
}

#[embassy_executor::task]
async fn run() {
    loop {
        esp_println::println!("Hello world from embassy using esp-hal-async!");
        Timer::after(Duration::from_millis(1_000)).await;
    }
}

#[main]
async fn main(spawner: Spawner) {
    esp_println::logger::init_logger_from_env();

    esp_println::println!("Init!");
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let timg0 = TimerGroup::new(peripherals.TIMG0, &clocks, None);
    let timer0 = OneShotTimer::new(timg0.timer0.into());
    let timers = [timer0];
    let timers = mk_static!([OneShotTimer<ErasedTimer>; 1], timers);
    esp_hal_embassy::init(&clocks, timers);

    spawner.spawn(run()).ok();

    loop {
        esp_println::println!("Bing!");
        Timer::after(Duration::from_millis(5_000)).await;
    }
}
```

but I keep getting:

```
❯ cargo b --release
   Compiling embedded-controller v0.1.0 (/home/skrapi/dev/treadmill/embedded-controller)
error[E0432]: unresolved import `esp_hal::timer::ErasedTimer`
  --> src/main.rs:12:31
   |
12 |     timer::{timg::TimerGroup, ErasedTimer, OneShotTimer},
   |                               ^^^^^^^^^^^ no `ErasedTimer` in `timer`

For more information about this error, try `rustc --explain E0432`.
error: could not compile `embedded-controller` (bin "embedded-controller") due to 1 previous error
```

I have scoured https://docs.esp-rs.org/esp-hal/esp-hal/0.18.0/esp32c3/esp_hal/

but don't see the option to include ErasedTimer. 

Any help would be appreciated.

skrapi, 2024-07-14:
Seems like  #1753 might have caused the difference

Dominaezzz, 2024-07-14:
> and then copied the text from the embassy_hello_world.rs example. I added the missing imports

Copy the example from the repo at the version you have used. There are often breaking changes between a released version and what's in the `main` branch.

skrapi, 2024-07-14:
good point, I tried that, solved my issue, but led to a separate one:

new src/main.rs
```
//! Blinks an LED
//!
//! This assumes that a LED is connected to the pin assigned to `led`. (GPIO0)

//% CHIPS: esp32 esp32c2 esp32c3 esp32c6 esp32h2 esp32s2 esp32s3

#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    delay::Delay,
    gpio::{Io, Level, Output},
    peripherals::Peripherals,
    prelude::*,
    system::SystemControl,
};

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Set GPIO0 as an output, and set its state high initially.
    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);
    let mut led = Output::new(io.pins.gpio7, Level::High);

    let delay = Delay::new(&clocks);

    loop {
        led.toggle();
        delay.delay_millis(500);
        led.toggle();
        // or using `fugit` duration
        delay.delay(2.secs());
    }
}
```

new error:
```
 cargo b --release
   Compiling embedded-controller v0.1.0 (/home/skrapi/dev/treadmill/embedded-controller)
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/skrapi/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/skrapi/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/skrapi/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/skrapi/.local/bin:/home/skrapi/.local/bin:/home/skrapi/.juliaup/bin:/home/skrapi/.nvm/versions/node/v20.9.0/bin:/home/skrapi/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustcyGGzsb/symbols.o" "/home/skrapi/dev/treadmill/embedded-controller/target/riscv32imc-unknown-none-elf/release/deps/embedded_controller-66ae9f1b14e2f9a4.embedded_controller.bfe8676fdc08d2b8-cgu.0.rcgu.o" "--as-needed" "-L" "/home/skrapi/dev/treadmill/embedded-controller/target/riscv32imc-unknown-none-elf/release/deps" "-L" "/home/skrapi/dev/treadmill/embedded-controller/target/release/deps" "-L" "/home/skrapi/dev/treadmill/embedded-controller/target/riscv32imc-unknown-none-elf/release/build/esp-hal-fb7671c3d3905182/out" "-L" "/home/skrapi/dev/treadmill/embedded-controller/target/riscv32imc-unknown-none-elf/release/build/esp32c3-e800466241b30004/out" "-L" "/home/skrapi/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/riscv32imc-unknown-none-elf/lib" "-Bstatic" "/home/skrapi/dev/treadmill/embedded-controller/target/riscv32imc-unknown-none-elf/release/deps/libcompiler_builtins-055f71e45e4d2fb5.rlib" "-Bdynamic" "-z" "noexecstack" "-L" "/home/skrapi/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/riscv32imc-unknown-none-elf/lib" "-o" "/home/skrapi/dev/treadmill/embedded-controller/target/riscv32imc-unknown-none-elf/release/deps/embedded_controller-66ae9f1b14e2f9a4" "--gc-sections" "-Tlinkall.x"
  = note: rust-lld: error: undefined symbol: _embassy_time_schedule_wake
          >>> referenced by mod.rs:1785 (/home/skrapi/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:1785)
          >>>               /home/skrapi/dev/treadmill/embedded-controller/target/riscv32imc-unknown-none-elf/release/deps/embedded_controller-66ae9f1b14e2f9a4.embedded_controller.bfe8676fdc08d2b8-cgu.0.rcgu.o:(_$LT$embassy_time..timer..Timer$u20$as$u20$core..future..future..Future$GT$::poll::hb20fcfd3bfb6b57b)
          

error: could not compile `embedded-controller` (bin "embedded-controller") due to 1 previous error
```

Anyway will close this issue soon, because the original error is not a real issue.

bjoernQ, 2024-07-15:
The linker error is can be resolved probably by using the feature `generic-queue-8` on `embassy-time` or `integrated-timers` on `esp-hal-embassy`

skrapi, 2024-07-15:
That worked, thanks folks

closed 2024-07-15 by skrapi