esp-rs/esp-hal #1798: Bugs in FlashSafeDma<SpiDma>

Dominaezzz, 2024-07-15:
Discovered these while refactoring the DMA APIs. (I'm going to be removing them anyway for the new DMA Move API but I thought it'd be worth mentioning in case someone else runs into it)

Here the result of the transfer isn't copied back into user provided buffer.
https://github.com/esp-rs/esp-hal/blob/7ea471c1ac2b7fbb4468a3f4211225e5dec2fe5c/esp-hal/src/spi/master.rs#L1687-L1700
https://github.com/esp-rs/esp-hal/blob/7ea471c1ac2b7fbb4468a3f4211225e5dec2fe5c/esp-hal/src/spi/master.rs#L1801-L1811

Here the DMA buffer isn't being used.
https://github.com/esp-rs/esp-hal/blob/7ea471c1ac2b7fbb4468a3f4211225e5dec2fe5c/esp-hal/src/spi/master.rs#L1482-L1491
https://github.com/esp-rs/esp-hal/blob/7ea471c1ac2b7fbb4468a3f4211225e5dec2fe5c/esp-hal/src/spi/master.rs#L1662-L1664
https://github.com/esp-rs/esp-hal/blob/7ea471c1ac2b7fbb4468a3f4211225e5dec2fe5c/esp-hal/src/spi/master.rs#L1683-L1685

plaes, 2024-07-24:
Two issues I noticed with current main (5d6354ccbd62527452968074c4a92a4fc6c6a635) as of 2024.07.24 when communicating using async from embassy task:
1. `FlashSafeDma` transfer hangs with certain buffer sizes: `3, 5, 6 and 8` words (tested up to 12 for now)

```
Hanging:
[INFO] - WRITE async FlashSafeDma: 0x3ffb0d5f, len: 3
[INFO] - WRITE async SpiDma: 0x3ffb0d5f, len: 3

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d5f, len: 5
[INFO] - WRITE async SpiDma: 0x3ffb0d5f, len: 5

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d5e, len: 6
[INFO] - WRITE async SpiDma: 0x3ffb0d5e, len: 6

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d58, len: 8
[INFO] - WRITE async SpiDma: 0x3ffb0d58, len: 8

```

```
Working:
[INFO] - WRITE async FlashSafeDma: 0x3ffb0eb4, len: 2
[INFO] - WRITE async SpiDma: 0x3ffb0eb4, len: 2

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d58, len: 4
[INFO] - WRITE async SpiDma: 0x3ffb0d58, len: 4

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d5f, len: 7
[INFO] - WRITE async SpiDma: 0x3ffb0d5f, len: 7

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d5f, len: 9
[INFO] - WRITE async SpiDma: 0x3ffb0d5f, len: 9

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d5e, len: 10
[INFO] - WRITE async SpiDma: 0x3ffb0d5e, len: 10

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d5f, len: 11
[INFO] - WRITE async SpiDma: 0x3ffb0d5f, len: 11

[INFO] - WRITE async FlashSafeDma: 0x3ffb0d58, len: 12
[INFO] - WRITE async SpiDma: 0x3ffb0d58, len: 12

```

Dominaezzz, 2024-07-24:
Got any code to reproduce that?

plaes, 2024-07-24:
> Got any code to reproduce that?

Well, repository of my LoRa code is currently here  - https://github.com/plaes/rust-lilygo-ttgo-lora32 which uses lora-phy to for communication with sx1276 utilizing `ExclusiveSpiDevice` and embedded-hal-1.0's transfer method:
https://github.com/lora-rs/lora-rs/blob/36885275fbfed98578e1a3a48555d973638b6105/lora-phy/src/interface.rs#L40-L42

PR #1856 (2024-08-20): [3/3] DMA Move API: Introduce DMA buffer objects
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** sec [... cut]