esp-rs/esp-hal #1811: `usb_serial_jtag` example freezes when pressing a key

bjoernQ, 2024-07-16:
While `embassy_usb_serial_jtag` works - `usb_serial_jtag` doesn't work for me. Tested on ESP32-S3, ESP32-C3, ESP32-C6 ... all the same.

JurajSadel, 2024-07-18:
I tried to run the `usb_serial_jtag` and `embassy_usb_serial_jtag` examples and both are working for me. I will try to run them on Windows later today as well

bjoernQ, 2024-07-18:
On Windows using espflash (3.1.0) this is 100% reproducible. However using putty it doesn't freeze for me.

Still the embassy example works so there is _something_ strange here

JurajSadel, 2024-07-19:
I've tried to run it on Windows, but it doesn't recognize any USB-SERIAL-JTAG port ... I followed steps from IDF and it still doesn't work.

bjoernQ, 2024-07-22:
This is most probably more of a problem in `espflash` than with HAL code. While the problem is seen when flashing and monitoring the problem isn't present when just flashing and the running `espflash monitor` separately

bjoernQ, 2024-07-22:
Recreating the serialport instance before monitoring might help - in any case this is definitely not a HAL issue but an espflash issue

closed 2024-07-22 by jessebraham