esp-rs/esp-hal #1817: esp-wifi 0.7 does not compile with `embedded-svc` feature

GnomedDev, 2024-07-16:
It seems like `esp-wifi` was never updated for https://github.com/esp-rs/embedded-svc/pull/75 when bumping the version number, and there is no CI coverage for `esp-wifi` with `embedded-svc`.

bjoernQ, 2024-07-17:
The good thing is embedded-svc is totally optional and we have an inherent WiFi API which looks the same - just needs different imports

With the (currently configured) version 0.27.1 it compiles fine.

Just updating the version to a semver incompatible version is expected to fail.

Nevertheless, updating embedded-svc might be a good thing. Also checking the feature in CI is probably something we want.

Will change the title of this issue accordingly

GnomedDev, 2024-07-17:
I don't know if I'm missing something, but esp-wifi is currently depending on embedded-svc 0.28 and currently fails to build with the embdded-svc feature enabled, contradicting "with the (currently configured) version 0.27 it compiles fine". https://github.com/esp-rs/esp-hal/blob/14baad4625b8699b61294e104a28e1c9b9dfb6b5/esp-wifi/Cargo.toml#L23-L24

This is blocking updating esp-mbedtls to the latest versions of the hal crates.

bjoernQ, 2024-07-17:
> ...  but esp-wifi is currently depending on embedded-svc 0.28 ...

Oh, sorry for that! I haven't noticed it was changed. You are right - the released 0.7.0 depends on 0.28.0

bjoernQ, 2024-07-17:
I will publish 0.7.1 soon

closed 2024-07-17 by bjoernQ

bjoernQ, 2024-07-17:
0.7.1 is published, 0.7.0 is yanked now. Thanks for reporting this issue

GnomedDev, 2024-07-17:
Should another issue be made for making sure this doesn't happen again (adding CI coverage for this feature)?

bjoernQ, 2024-07-17:
> Should another issue be made for making sure this doesn't happen again (adding CI coverage for this feature)?

I think #1693 remotely touches the issue - probably worth a comment there to not forget about this (especially since that issue is planned to be worked on for the upcoming milestone)

PR #1820 (2024-07-17): Prepare esp-wifi 0.7.1
This prepares a hot-fix for esp-wifi

Closes #1817