esp-rs/esp-hal #1826: Expose SPI sio() setting

ElDonad, 2024-07-17:
Hello,
I was trying to setup three wires SPI for my application using an ESP32-S3, and it appears that the esp-hal API doesn't expose this setting. Would it be possible to expose this functionality ? I don't think it would be very hard, there could be a helper function akin to `Spi::with_sio()` that would resemble something like 
```rust
fun with_sio(self) -> Self{
    let reg_block = self.register_block();
    reg_block.user().modify(|_,w| unsafe {
        w.sio().set_bit();
    }
    self
}
```

I'm fairly new to ESP and Rust dev and I don't have the means to thoroughly test this PoC currently, so I hope it's not too far from correct...

bjoernQ, 2024-07-18:
That would be a useful addition

ElDonad, 2024-07-22:
By the way there is not much information about it in the technical manual, but I assume this should only be available for the Half Duplex mode, it wouldn't make sense otherwise. I'll try to get it to work on a ESP32S3 I have, maybe then I can issue a PR...

PR #2919 (2025-01-17): Support 3-wire-SPI
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I have added neces [... cut]
closed 2025-01-17 by MabezDev