esp-rs/esp-hal #1843: Improving the logging performance of `esp-println` with async TX

ProfFan, 2024-07-22:
The current implementation of `esp-println` is blocking (which is totally reasonable) but the performance is not great as it blocks everything else in a critical region.

In my project I designed an async version that TX in a separate embassy task. This improves the performance of logging 10x at least, and allows `esp-wifi` with TRACE enabled to run fast.

I am thinking about porting this to `esp-println`. However currently this only works with `embassy` so I need suggestions on how to achieve this without it.

My implementation uses a custom halt function in `esp-backtrace` to ensure that the panic message is flushed to the USB serial JTAG in case of even catastrophic failure.

My implementation lives in https://github.com/ProfFan/magic-loc-ng/blob/7280f7e4a6b0062bf56595aa8017f035fbcd316c/src/serial/mod.rs

bjoernQ, 2024-07-23:
This sounds like a great thing for a 3rd party component I guess

MabezDev, 2024-08-21:
This is very cool! I think that we want to keep esp-println and friends as simple as possible though. As @bjoernQ mentioned, this would be great as a 3rd party alternative.

closed 2024-08-21 by MabezDev