esp-rs/esp-hal #1857: Async SPI broken for ESP32

bjoernQ, 2024-07-25:
Since 4ef91c5941883a5dc589209eb91fe25641570686 Async-SPI is broken for ESP32

Steps to reproduce:
- connect GPIO 2 and 4
- run `cargo xtask run-example examples esp32 embassy_spi`

Expected:
It works as before

Actual:
`transfer` never returns

bjoernQ, 2024-07-25:
(Still works for ESP32-S2)

Dominaezzz, 2024-07-25:
I reckon this is due to the interrupts being cleared after the DMA transfer starts, I'll have time to investigate this evening and I'll add a hil test for async spi

plaes, 2024-07-25:
Another testcase which is broken even with the DMA patch:
```rust
    let mut write_buf = &[0xaa];
    //let mut payload = &[0xd1, 0xd2, 0xd3, 0xd4];
    // Works with 1..4 bytes
    // Hangs when payload is > 4 bytes??
    let mut payload = &[0xd1, 0xd2, 0xd3, 0xd5, 0xd5];

    defmt::info!("TX... ");
    let mut ops = [Operation::Write(write_buf), Operation::Write(payload)];
    let _ = spi_dev.transaction(&mut ops).await;
    defmt::info!("...done");
```

Dominaezzz, 2024-07-25:
Ah, unfortunately the HIL tests don't run on the ESP32... oh well, we'll have to settle for the example then.

PR #1859 (2024-07-26): SPI: Clear DMA interrupts before (not after) DMA starts
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper [... cut]
closed 2024-07-26 by bjoernQ