esp-rs/esp-hal #1858: Provide some way to *not* bind an interrupt handler in `Io::new`

jessebraham, 2024-07-25:
There are cases where users may want to use the GPIO interrupt in ways other than what is currently supported, since we are setting an interrupt handler by default:  
https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/gpio/mod.rs#L1192-L1216

It would be nice to have some way to indicate that this handler should not be set.

cc @bjoernQ @MabezDev

bjoernQ, 2024-07-26:
Easiest would be to have another constructor for this very special case - with a big warning in the docs regarding async etc. - I guess this corner case isn't worth to add type-state

jessebraham, 2024-07-26:
I agree, this is probably the simplest path forward. Just wanted to run it past you first as you wrote most of this code in the first place 😁 Thanks for the input!

PR #1861 (2024-07-26): Add a new constructor to `gpio::Io` which does not bind an interrupt handler
Wasn't sure what to call it (as usual), but I think this is a reasonable enough name. Happy to change it if anybody has any better suggestions.

I think the comment should hopefully suffice, but again let me know if I should make it more scary and/or mention anything else.

Closes #1858
closed 2024-07-26 by MabezDev