esp-rs/esp-hal #1865: Should we remove the `embedded-hal-nb` trait implementations?

jessebraham, 2024-07-27:
I checked the dependent crates for this package and almost all of them are HALs, there were maybe a single-digit number of device drivers which actually implemented this traits. None of these drivers seemed to have an especially large number of downloads.

I never personally hear of or see people using these trait implementations.

What are we gaining by keeping these trait implementations? Or are we just carrying around extra baggage for the sake of it?

bjoernQ, 2024-07-29:
I agree they are probably not super useful these days. On the other hand, I _think_ they don't add a lot of maintenance burden and even `embassy_stm32` still implements the traits

But t.b.h. I don't have a strong opinion on this

ProfFan, 2024-07-29:
They are good for non-async code that use event loops, as async and sync are like two worlds in Rust. Especially for DMA I think.

MabezDev, 2024-08-08:
Imo there is no harm in keeping them, if they start to become a maintenance burden we can revisit this decision.

closed 2024-08-08 by MabezDev