esp-rs/esp-hal #1866: Cannot build esp-wifi. toml_cfg errors

therealchfkch, 2024-07-27:
As is was cloning the examples in the training from https://github.com/esp-rs/no_std-training I noticed the build fails with the following errors https://pastebin.com/MdC9vMFC.

Short form of some symptomatic error messages here:
```
Compiling esp-wifi v0.7.1
error[E0463]: can't find crate for `toml_cfg`
  --> /home/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.7.1/src/lib.rs:72:3
   |
72 | #[toml_cfg::toml_config]
   |   ^^^^^^^^ can't find crate
 
error: cannot determine resolution for the attribute macro `toml_cfg::toml_config`
  --> /home/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.7.1/src/lib.rs:72:3
   |
72 | #[toml_cfg::toml_config]
   |   ^^^^^^^^^^^^^^^^^^^^^
   |
   = note: import resolution is stuck, try simplifying macro imports
 
error: cannot find attribute `default` in this scope
  --> /home/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.7.1/src/lib.rs:75:7
   |
75 |     #[default(5)]
   |       ^^^^^^^

rror[E0425]: cannot find value `CONFIG` in the crate root
   --> /home/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.7.1/src/wifi/mod.rs:664:37
    |
664 | const RX_QUEUE_SIZE: usize = crate::CONFIG.rx_queue_size;
    |                                     ^^^^^^ not found in the crate root
 
error[E0425]: cannot find value `CONFIG` in the crate root
   --> /home/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.7.1/src/wifi/mod.rs:665:37
    |
665 | const TX_QUEUE_SIZE: usize = crate::CONFIG.tx_queue_size;
    |                                     ^^^^^^ not found in the crate root
 
error[E0425]: cannot find value `CONFIG` in the crate root
    --> /home/xxxxx/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.7.1/src/wifi/mod.rs:1064:31
     |
1064 |     static_rx_buf_num: crate::CONFIG.static_rx_buf_num as i32,
     |                               ^^^^^^ not found in the crate root
```

I also tried using the `cargo-generate` template, which produces the same output.

**Board used:** ESP32C6 (but i also tried just to build with C3 board configured)
**Channel:** nightly and nightly-2023-11-14 (as it was a preset in the example.

I suspect somehow the build dependency is not properly included, but it is declared here: https://github.com/esp-rs/esp-hal/blob/237804efb602fe22542be36c90c919efc14879fd/esp-wifi/Cargo.toml#L56

Am i doing something wrong while building? I followed all the books and tutorials i found. Everything else works as suspected, except when i include `esp-wifi` in the project. "Normal" projects work on both C6 and C3 boards previously.

bjoernQ, 2024-07-29:
Most probably something messed up you Cargo cache - you could use e.g. `cargo-cache` to clean-up or manually clean things in `$HOME/.cargo`

therealchfkch, 2024-07-29:
I did now both clear with cargo-cache and manually, but the outcome is still the same. Can you confirm that you can build the `wifi_embassy_dhcp` or `wifi_dhcp`example? So i can be sure it is a problem on my end and not the upstream repo.

bjoernQ, 2024-07-29:
Yes, it builds fine. We also build all examples in CI so we should always be sure all examples are always buildable

<details>
  <summary>Build log</summary>
  
```
❯ cargo xtask run-example examples esp32c6 wifi_embassy_dhcp
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.09s
     Running `target\debug\xtask.exe run-example examples esp32c6 wifi_embassy_dhcp`
[2024-07-29T07:50:42Z INFO  xtask] Building example 'C:\projects\esp\esp-hal\examples\src\bin\wifi_embassy_dhcp.rs' for 'esp32c6'
[2024-07-29T07:50:42Z INFO  xtask]   Features: async,embassy,embassy-generic-timers,esp-wifi,esp-wifi/async,esp-wifi/embassy-net,esp-wifi/wifi-default,esp-wifi/wifi,esp-wifi/utils
[2024-07-29T07:50:42Z INFO  xtask] Package: "src\\bin\\wifi_embassy_dhcp.rs"
   Compiling compiler_builtins v0.1.109
   Compiling core v0.0.0 (C:\Users\Bjoern\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core)
   Compiling proc-macro2 v1.0.86
   Compiling unicode-ident v1.0.12
   Compiling version_check v0.9.4
   Compiling windows_x86_64_msvc v0.52.6
   Compiling typenum v1.17.0
   Compiling portable-atomic v1.6.0
   Compiling strsim v0.11.1
   Compiling serde v1.0.204
   Compiling rustversion v1.0.17
   Compiling anyhow v1.0.86
   Compiling heck v0.5.0
   Compiling cfg-if v1.0.0
   Compiling utf8parse v0.2.2
   Compiling anstyle v1.0.7
   Compiling windows-targets v0.52.6
   Compiling windows-sys v0.52.0
   Compiling anstyle-parse v0.2.4
   Compiling generic-array v0.14.7
   Compiling heapless v0.8.0
   Compiling syn v1.0.109
   Compiling is_terminal_polyfill v1.70.0
   Compiling colorchoice v1.0.1
   Compiling fnv v1.0.7
   Compiling ident_case v1.0.1
   Compiling clap_lex v0.7.1
   Compiling equivalent v1.0.1
   Compiling hashbrown v0.14.5
   Compiling embedded-io-async v0.6.1
   Compiling lazy_static v1.5.0
   Compiling proc-macro-error-attr v1.0.4
   Compiling quote v1.0.36
   Compiling litrs v0.4.1
   Compiling proc-macro-error v1.0.4
   Compiling embedded-hal-async v1.0.0
   Compiling embassy-sync v0.5.0
   Compiling syn v2.0.71
   Compiling static_assertions v1.1.0
   Compiling semver v1.0.23
   Compiling indexmap v2.2.6
   Compiling adler v1.0.2
   Compiling twox-hash v1.6.3
   Compiling crc32fast v1.4.2
   Compiling miniz_oxide v0.7.4
   Compiling winapi-util v0.1.8
   Compiling anstyle-wincon v3.0.3
   Compiling anstyle-query v1.1.0
   Compiling termcolor v1.4.1
   Compiling anstream v0.6.14
   Compiling winnow v0.5.40
   Compiling riscv v0.11.1
   Compiling document-features v0.2.10
   Compiling byteorder v1.5.0
   Compiling flate2 v1.0.30
   Compiling ruzstd v0.7.0
   Compiling clap_builder v4.5.10
   Compiling embassy-sync v0.6.0
[... 144 lines cut ...]
   Compiling num_enum v0.7.2
   Compiling micromath v2.1.0
   Compiling lis3dh-async v0.9.3
   Compiling bleps v0.1.0 (https://github.com/bjoernQ/bleps?rev=a5148d8ae679e021b78f53fd33afb8bb35d0b62e#a5148d8a)
   Compiling usbd-serial v0.2.2
   Compiling embedded-graphics v0.8.1
   Compiling aes v0.8.4
   Compiling p192 v0.13.0
   Compiling p256 v0.13.2
   Compiling ssd1306 v0.8.4
   Compiling embassy-net v0.4.0
   Compiling esp-alloc v0.4.0 (C:\projects\esp\esp-hal\esp-alloc)
   Compiling toml-cfg v0.2.0
   Compiling sha2 v0.10.8
   Compiling hmac v0.12.1
   Compiling smart-leds v0.4.0
   Compiling static_cell v2.1.0
   Compiling embedded-storage v0.3.1
   Compiling hex-literal v0.4.1
   Compiling esp-wifi v0.7.1 (C:\projects\esp\esp-hal\esp-wifi)
   Compiling esp-hal-smartled v0.12.0 (C:\projects\esp\esp-hal\esp-hal-smartled)
    Finished `release` profile [optimized + debuginfo] target(s) in 53.73s
     Running `espflash flash --monitor target\riscv32imac-unknown-none-elf\release\wifi_embassy_dhcp`
```
</details>

therealchfkch, 2024-07-29:
Okay, thanks so far. Maybe i will check it out on another machine then.

therealchfkch, 2024-07-29:
Hhm on my Alpine/PMOS phone it builds, seems to be a NixOS or configuration problem on my side then. Should i close it or leave it open for discussion?

bjoernQ, 2024-07-29:
I'll close this for now but feel free to add any relevant information so anyone else running into a similar issue will be able to find additional information here. Thanks

closed 2024-07-29 by bjoernQ