esp-rs/esp-hal #1869: /esp-hal/src/timer/timg.rs examples do not work for TimerGroup

bbustin, 2024-07-28:
timg::TimerGroup now requires an additional argument with a Wdt<T, DM>. The documentation in the code shows only two arguments and thus does not work.

JurajSadel, 2024-07-29:
Hello @bbustin, thank you for reporting the issue! I'm not sure if I understand correctly what the issue is - could you please: 
1. link the example that does't work for you
2. which chip are you using
3. what esp-hal version are you using?

Thanks!

bbustin, 2024-07-29:
I am using esp-hal version 0.19.0 with feature esp32c6.

    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);

    let clocks = ClockControl::max(system.clock_control).freeze();
    let delay = Delay::new(&clocks);

    let timg0 = TimerGroup::new(peripherals.TIMG0, &clocks);

The last line gives the error "this function takes 3 arguments, but 2 arguments were supplied". It looks like TimerGroup::new was modified to need an additional parameter of type Wdt.

Dominaezzz, 2024-07-29:
See https://github.com/esp-rs/esp-hal/blob/v0.19.0/esp-hal/src/timer/timg.rs#L241

`let timg0 = TimerGroup::new(peripherals.TIMG0, &clocks, None);`

bbustin, 2024-07-29:
Thanks @JurajSadel and @Dominaezzz. This looks like a [PEBKAC](https://en.wikipedia.org/wiki/User_error) error. I apologize. I really thought I'd found an issue. Now, after @Dominaezzz pointed it out, it was correct all along. I must have subconsciously ignored the None.

I apologize for wasting your time. Thanks for this great project. I really appreciate all your hard work.

closed 2024-07-29 by bbustin