esp-rs/esp-hal #1878: Cargo.toml example for embassy_i2c.rs

marvinrobot42, 2024-07-29:
Does anyone have a Cargo.toml file using the ESP32 C6 (or C3) for the embassy_i2c.rs example.  I am stuck trying to update dependency esp_hal to version 0.19.0 which appears to have the embassy feature removed.

If I add the esp-hal-embassy crate I get this build error:
#[esp_hal_embassy::main]
   |                    ^^^^ could not find `main` in `esp_hal_embassy`

Also, it looks like the let timg0 = TimerGroup::new(  ...  statement is missing an 'isr' argument.  I wonder how that is created.

bjoernQ, 2024-07-30:
Generally, the `main` branch is the "next"/WIP version - when looking into examples for a released version it's better to look at the corresponding tag: https://github.com/esp-rs/esp-hal/tree/v0.19.0

marvinrobot42, 2024-07-30:
Thanks, the v0.19.0 release branch examples helpful.  Problem solved.