esp-rs/esp-hal #1885: ESP32-S2: `systimer` example shows different behavior than before and on other chips

bjoernQ, 2024-07-31:
First commit where I noticed it is 8e0a2012dcc92efcfc71ac18c51e9809cb59c929 - this doesn't mean it's the offending commit - needs a proper git-bisect

Before, the example quickly starts to print from the ISRs - now it doesn't.

I don't think there is anything fundamentally broken since esp-wifi and embassy time-driver still works fine - nevertheless we should do something about this (probably low-prio)

playfulFence, 2024-08-08:
This is what `bisect` told me: 
```shell
4f9dc960c652ab7a4652cb9e6b21fc75cfaf32fd is the first bad commit
commit 4f9dc960c652ab7a4652cb9e6b21fc75cfaf32fd
Author: Björn Quentin <bjoernQ@users.noreply.github.com>
Date:   Wed Jul 3 17:43:01 2024 +0200

    Improve xtensa interrupt latency (#1735)
    
    * Avoid U128
    
    * Avoid unnecessary calls
    
    * CHANGELOG.md
    
    * Simplify
    
    Co-authored-by: Jesse Braham <jessebraham@users.noreply.github.com>
    
    ---------
    
    Co-authored-by: Jesse Braham <jessebraham@users.noreply.github.com>
 ```
 
 see https://github.com/esp-rs/esp-hal/pull/1735
 
 After some more test I can confirm that before this commit `systimer` worked as expected.

playfulFence, 2024-08-08:
Upd:
Even one commit before this it was hanging in this state: 
[image]

Will do some further investigation

bugadani, 2024-08-20:
Oh fun, I want to give this one a try if someone doesn't figure it out until tomorrow :D At the very least, this should be a simple enough test case, too, to catch potential regressions.