esp-rs/esp-hal #1905: Implement touch support for `esp32s2` and `esp32s3`.

jounathaen, 2024-08-05:
As of https://github.com/esp-rs/esp-hal/pull/1873, esp-hal only has touch support for the `esp32`. However, the `esp32s2` and `esp32s3` do also have touch peripherals.

The `esp32s3`'s peripheral has a rather similar structure as the `esp32`'s one, but judging from the datasheet, the `esp32s2`'s looks a little more different.

To get a starting point, I made have a very rough, incomplete and not working first attempt for the `esp32s3`, but at least it includes the pin definitions: https://github.com/jounathaen/esp-hal/tree/touch-esp32s3 
(As I don't have any hardware, (and honestly, also no motivating use-case), I will not continue with the implementation myself.)

playfulFence, 2024-08-19:
I'm taking `esp32s2`.

nponsard, 2025-01-05:
Hello, I am trying to add touch support for the `esp32s3`.
It seems like the file `esp-hal/src/touch.rs` is attempting to read/write to registers named differently between the `esp32` and `esp32s3`.
Where does board specific code need to be moved ? Is it the `esp-hal/src/soc` folders ?

bugadani, 2025-01-06:
> Where does board specific code need to be moved ? Is it the esp-hal/src/soc folders ?

That sounds like a good place for now, yes.

nponsard, 2025-02-10:
From what I read in the esp-idf source code, esp32s2 and esp32s3 uses the same 'version 2' touch hardware (from the comment in [touch_version_types.h](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_touch_sens/hw_ver2/include/driver/touch_version_types.h)).

On my side I tried to loosely adapt the existing code for the esp32 to use the registers for the esp32s3, but it currently doesn't work. 

I probably need to do some tests without using the traits and structs already there for the esp32, so I can understand how to properly initialize everything.

If you want to take a look at my code (that is not working) it is available here https://github.com/nponsard/esp-hal/tree/feat/esp32s3-touch.

nponsard, 2025-03-31:
> I looked at the code but the init phase and how to sample the pins is very different from the esp32 v1. This code as you mention you be different. Did you already have a working example without trying to implement the traits? Let me know, will give a try too. Will look at the c++/idf implementatiiom to get inspired.

Hello, I uploaded my test code here [https://github.com/nponsard/esp32s3-touch-test-](https://github.com/nponsard/esp32s3-touch-test-), you have to use my branch for it to compile. 

The initialization follows all the steps of the idf code, but I havent managed to get a good read.

foldingbeauty, 2025-08-19:
@nponsard did you update your code with the right test?. I have more time to spend after the summer break and I want to take another look at your code and see if we can get a good read. Let me know if you have an update or use the code as-is to make it work.

nponsard, 2025-08-19:
> [@nponsard](https://github.com/nponsard) did you update your code with the right test?. I have more time to spend after the summer break and I want to take another look at your code and see if we can get a good read. Let me know if you have an update or use the code as-is to make it work.

I'm sorry I couldn't find the project I was referring to. My fork seems to have all the initialization bits I remember writing so maybe everything is already there.

foldingbeauty, 2025-11-02:
update. I took @nponsard code  and created a new fork to find out why there is no clean read. you can find my work (in progress) here https://github.com/hi-squeaky-things/esp-hal 

I will start with the one-shot mode and I already can see the FSM of the touch device is not trigger, will focus on this first, also the PAC has some interesting nummering of the pads, which I already fixed in my code. The PAC is fine but you need to call the right touch_conf register.

Keep you all updated. I also got myself an esp32s2 so I can test it on a single xtensa cpu too.

Keep you updated on the progress. I will also test it against my own reference hardware platform (which contains a touch matrix):

https://github.com/hi-squeaky-things/little-squeaky-machine-hardware