esp-rs/esp-hal #1909: I2C slave driver

okhsunrog, 2024-08-07:
Are there any plans for I2C slave driver? I only found embassy example for I2C slave for rp2040 https://github.com/embassy-rs/embassy/blob/main/examples/rp/src/bin/i2c_slave.rs
Checked the i2c driver in esp-hal and it seems like there isn't any slave driver.

jessebraham, 2024-08-07:
While we would like to have this eventually, we have no plans to work on this feature at this point in time. Likely this will need to be a community contribution, but perhaps one of the maintainers will have time in the coming months if not.

If this feature generates enough interest we can definitely re-evaluate and prioritize this, but you are one of (if not *this*) first people to ask :)

okhsunrog, 2024-08-07:
Okay. Maybe I'll open a draft PR in the next month if I have free time 😅

marcpabst, 2025-01-22:
I was a bit surprised to see there’s no I2C slave driver in esp-hal. This would be a great addition!

hcabel, 2025-02-05:
I agree, +1 for i2c slave driver

turgu1, 2025-10-22:
I decided to work on a I2C slave driver. At this point in time, it works fine for the esp32C6. As I have in hands both ESP32 and ESP32S3 devices, I will complete the tests on these devices before making it available as a PR. You can find it [here](https://github.com/turgu1/esp-hal/tree/i2c-slave-driver) in folder `esp-hal/esp-hal/src/i2c/slave`. The effort is being done with the help of Claude Sonnet 4.5 .

turgu1, 2025-11-30:
@mbehr1 : Hello Matthias, I've been away from that code for the last 3 weeks and I'm not expecting to look at it before January. Here are my idea on that:
- Is it possible for the master to try to read a larger size of data even if it receives less amount? Normally the slave would indicates the end of the transmission end the master would then be able to access the data. A possible issue could be that the i2c fifo on the slave side needs to be cleared before writing again.
- For the async issue, I have been trying to get it work without success with the i2c/slave folder version. A bit normal as the I2C is synchronous by definition. I started to work on a second interrupt based version but it still not in a working state. You may find it in the i2c/slave_async folder.

mbehr1, 2025-12-27:
@turgu1 see e.g. here for an example usage: https://github.com/mbehr1/esp-junkers-bm1/blob/fa861954093b2fc4be905dad34c633a7c3fc83fd/src/i2c.rs#L84

wallem89, 2026-02-23:
I would love to see one of these solution to be merged soon! Can I contribute in anyway to speed this up? I have some different devkits laying around for testing. 

For me a valid use case is to simulate or mock behavior of an I2C slave device with an ESP32.