esp-rs/esp-hal #1915: xtensa-lx feature not found in esp-hal

lelongg, 2024-08-09:
I can't build the project generated by _esp-rs/esp-template_ for esp32s3 with target set to _xtensa-esp32s3-none-elf_.

```
error: failed to select a version for `esp-hal`.
    ... required by package `esp-now-demo v0.1.0 (/home/sisyphe/Projects/third_party/esp-now-demo)`
versions that meet the requirements `*` (locked to 0.19.0) are: 0.19.0

the package `esp-now-demo` depends on `esp-hal`, with features: `xtensa-lx` but `esp-hal` does not have these features.
 It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.

failed to select a version for `esp-hal` which could resolve this conflict
```

bjoernQ, 2024-08-12:
I just tried the current template and the resulting project was buildable for me. Mind sharing the exact steps (i.e. options you used, modifications you did to the generated project)?

MauroMombelli, 2024-08-15:
same issue here:

> cargo generate esp-rs/esp-template                                                                                                                                                                                                                                                     
⚠   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
🤷   Project Name: dshot-rs
🔧   Destination: ~/dshot-rs ...
🔧   project-name: dshot-rs ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32
✔ 🤷   Configure advanced template options? · false
✔ 🤷   The template is requesting to run the following command. Do you agree?
cargo fmt · yes
🔧   Moving generated files into: `~/dshot-rs`...
Initializing a fresh Git repository
✨   Done! New project created ~/dshot-rs

THEN inside of the folder

> cargo build
    Updating crates.io index
error: failed to select a version for `esp-hal`.
    ... required by package `dshot-rs v0.1.0 (~/dshot-rs)`
versions that meet the requirements `^0.19.0` are: 0.19.0

the package `dshot-rs` depends on `esp-hal`, with features: `xtensa-lx` but `esp-hal` does not have these features.

MabezDev, 2024-08-16:
> same issue here:
> 
> ```
> > cargo generate esp-rs/esp-template                                                                                                                                                                                                                                                     
> ⚠   Favorite `esp-rs/esp-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-template.git
> 🤷   Project Name: dshot-rs
> 🔧   Destination: ~/dshot-rs ...
> 🔧   project-name: dshot-rs ...
> 🔧   Generating template ...
> ✔ 🤷   Which MCU to target? · esp32
> ✔ 🤷   Configure advanced template options? · false
> ✔ 🤷   The template is requesting to run the following command. Do you agree?
> cargo fmt · yes
> 🔧   Moving generated files into: `~/dshot-rs`...
> Initializing a fresh Git repository
> ✨   Done! New project created ~/dshot-rs
> ```
> 
> THEN inside of the folder
> 
> ```
> > cargo build
>     Updating crates.io index
> error: failed to select a version for `esp-hal`.
>     ... required by package `dshot-rs v0.1.0 (~/dshot-rs)`
> versions that meet the requirements `^0.19.0` are: 0.19.0
> 
> the package `dshot-rs` depends on `esp-hal`, with features: `xtensa-lx` but `esp-hal` does not have these features.
>  It has an optional dependency with that name, but that dependency uses the "dep:" syntax in the features table, so it does not have an implicit feature with that name.
> 
> 
> failed to select a version for `esp-hal` which could resolve this conflict
> ```

I followed your repro instructions, but on my machine it builds fine.

My env:
```
rustc +esp --version --verbose
rustc 1.80.0-nightly (1b259e0b8 2024-07-22) (1.80.0.0)
binary: rustc
commit-hash: 1b259e0b8b2129dd3d62819a32f6d9b287e6b86c
commit-date: 2024-07-22
host: aarch64-apple-darwin
release: 1.80.0-nightly
LLVM version: 17.0.1
``` 

Could you post the generated `Cargo.toml` too?

MauroMombelli, 2024-08-16:
```
[package]
name = "dshot-rs"
version = "0.1.0"
authors = ["mauro"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
esp-backtrace = { version = "0.13.0", features = [
    "esp32",
    "exception-handler",
    "panic-handler",
    "println",
] }
esp-hal = { version = "0.19.0", features = [ "esp32" ] }
esp-println = { version = "0.10.0", features = ["esp32", "log"] }
log = { version = "0.4.21" }
[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1 # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false
```

closed 2024-08-19 by jessebraham