esp-rs/esp-hal #1916: i2c async broken in #1900

liebman, 2024-08-09:
Async i2c write_read() hangs after #1900.  

This small program reads the seconds register from a DS3231 on pins sda:40 scl:41 once a second.

```rust
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    gpio::Io,
    i2c::I2C,
    peripherals::Peripherals,
    prelude::*,
    system::SystemControl,
    timer::{timg::TimerGroup, ErasedTimer, OneShotTimer},
};
use esp_println::println;

macro_rules! mk_static {
    ($t:ty,$val:expr) => {{
        static STATIC_CELL: static_cell::StaticCell<$t> = static_cell::StaticCell::new();
        #[deny(unused_attributes)]
        let x = STATIC_CELL.uninit().write(($val));
        x
    }};
}

#[esp_hal_embassy::main]
async fn main(_spawner: Spawner) {
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let timg0 = TimerGroup::new(peripherals.TIMG0, &clocks);
    let timer0: ErasedTimer = timg0.timer0.into();
    let timers = [OneShotTimer::new(timer0)];
    let timers = mk_static!([OneShotTimer<ErasedTimer>; 1], timers);
    esp_hal_embassy::init(&clocks, timers);

    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);

    let mut i2c0 = I2C::new_async(
        peripherals.I2C0,
        io.pins.gpio40,
        io.pins.gpio41,
        400.kHz(),
        &clocks,
    );

    let mut data = [0u8; 1];
    loop {
        println!("Reading (write_read) 1 byte from I2C device...");
        i2c0.write_read(0x68, &[0x00], &mut data).await.unwrap();
        println!("Read: {:02x?}", data[0]);
        Timer::after(Duration::from_millis(1000)).await;
    }
}
```

jessebraham, 2024-08-09:
cc @playfulFence

ProfFan, 2024-08-10:
From what I see I2C does not have a HIL test, this should be fixed ASAP

Szybet, 2024-08-13:
The same thing happens on esp32c6 :(

playfulFence, 2024-08-13:
Yes, I've already fixed that, PR will be published in next 30 minutes

closed 2024-08-13 by bugadani