esp-rs/esp-hal #1917: Make AnyPin accessible from `::gpio::AnyPin`?

plaes, 2024-08-10:
Currently import path for `AnyPin` for esp-hal is `::gpio::any_pin::AnyPin`. Both embassy-nrf and embassy-stm32 have AnyPin under `::gpio::AnyPin`.
Would it make sense to switch to `::gpio::AnyPin` for esp-hal as well?

bjoernQ, 2024-08-12:
Not saying we shouldn't but re-exporting the types doesn't look too nice in the docs:

[image]

If we think this improves the user experience we might want to move the code from the module into the super-module - originally, I wanted to have it separated because `gpio.rs` is already quite large

However no strong opinion on this - just something to consider

plaes, 2024-08-12:
Hmm.. need to check how `#[doc(inline)]` affects this.

plaes, 2024-08-13:
> Not saying we shouldn't but re-exporting the types doesn't look too nice in the docs: [...]

I seem to managed to fix it by hiding the documentation for `any_pin` submodule via `#doc(hidden)`. I still kept it public to not break existing applications, but it doesn't show up in documentation anymore.
Unfortunately there doesn't seem to be a way to mark a certain import path as deprecated.

closed 2024-08-26 by MabezDev