esp-rs/esp-hal #1920: Feature Request: Sleep and wake support for ESP32C2

M4tsuri, 2024-08-11:
Hello, I found that this crate does not provide sleep mode support for ESP32C2 while [the TRM](https://www.espressif.com/sites/default/files/documentation/esp8684_technical_reference_manual_en.pdf) states it is actually supported on hardware. I wonder if there is a specific reason or I'd like to handle this work if possible. Thank you.

jessebraham, 2024-08-12:
Support for light- and deep-sleep were originally added via a community contributor, and as such it had limited coverage. As @bjoernQ said, the only real reason this functionality does not exist currently is because nobody has implemented it, the ESP32-C2 is relatively low-priority for us compared with some other targets right now.

We would of course love to see a PR for this, but I understand this is not a small or simple feature so that may not be practical/possible. However, if more people show interest in this then we can perhaps prioritize it in the future.

bugadani, 2024-09-05:
Closed by #1922 I guess? Thank you for the PR!

closed 2024-09-05 by bugadani