esp-rs/esp-hal #1928: SD/MMC driver

Dominaezzz, 2024-08-12:
This would be great for talking to SD cards at faster speeds and also supporting esp-hosted on the ESP32-P4.

I have already started working on it (for the S3), so this is just a place holder issue so no one else duplicates my work.
I've managed to read from an SD Card in 1-bit mode using [`embedded-sdmmc`](https://github.com/rust-embedded-community/embedded-sdmmc-rs). Just need to add streaming, DMA and async support.

Regarding HIL testing, would it be alright to setup a devkit with an SD card on it (ideally exposing the pins for 4 bit mode)? I can't think of any other way to HIL test the driver.

MabezDev, 2024-08-12:
Feel free to take https://github.com/MabezDev/embedded-fatfs for a spin too :D. I've included a SPI SD card driver in there, but having proper SDIO would be awesome.

Regarding HIL, we should be able to hook up a read SD Card, as I think we have test infra for this in esp-idf (cc: @JurajSadel , @playfulFence )

ost-ing, 2024-12-23:
4bit Sdio/sdmmc support is pretty crucial for any serious application, id love to see support for this feature in the near future