esp-rs/esp-hal #1932: esp_wifi does not compile with esp_ieee802154 imported at the same time

turulix, 2024-08-13:
Hey im trying to use both WIFI and IEEE802154 at the same time on my ESP32C6.

```rs
#[global_allocator]
static ALLOCATOR: esp_alloc::EspHeap = esp_alloc::EspHeap::empty();

fn init_heap() {
    const HEAP_SIZE: usize = 32 * 1024;
    static mut HEAP: MaybeUninit<[u8; HEAP_SIZE]> = MaybeUninit::uninit();

    unsafe {
        ALLOCATOR.init(HEAP.as_mut_ptr() as *mut u8, HEAP_SIZE);
    }
}

#[entry]
fn main() -> ! {
    let mut peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    init_heap();
    esp_println::logger::init_logger_from_env();

    let clocks = ClockControl::max(system.clock_control).freeze();
    let rng = Rng::new(peripherals.RNG);
    let timer = TimerGroup::new(peripherals.TIMG0, &clocks, None);
    let timer: ErasedTimer = timer.timer0.into();
    let timer = PeriodicTimer::new(timer);

    let mut ieee802154 = esp_ieee802154::Ieee802154::new(peripherals.IEEE802154, &mut peripherals.RADIO_CLK);
    let init = esp_wifi::initialize(
        EspWifiInitFor::Wifi,
        timer,
        rng.clone(),
        peripherals.RADIO_CLK,
        &clocks,
    ).unwrap();

    loop {}
}
```

Commenting out either the esp_wifi part or ieee802154 part and removing all imports associated with them works fine. But compiling with both in place results in an `error: failed to load bitcode of module "esp_wifi-715e490f8c6e0fb1.esp_wifi.abe10e676f975f83-cgu.0.rcgu.o":` error

bjoernQ, 2024-08-13:
WiFi and 802.15.4 co-existence is not supported - while it is on esp-idf even there it's not recommended (e.g. that's why the ESP Thread Border Router is ESP32-S3 plus ESP32-H2)

However, both should be useable (but not at the same time) - e.g. for Matter commissioning via BLE and then using Thread

MabezDev, 2024-09-16:
@bjoernQ should the 802154 driver take RADIO_CLK too/instead to ensure this then?

bjoernQ, 2024-09-18:
Mhhh both drivers take RADIO_CLK already so it shouldn't be possible to even get to the linking stage

> So the linker problem here is that both esp-wifi and IEEE, define things like rtc_printf etc. @bjoernQ I think we can just move these to esp-wifi-sys which both crates depend on?

Independently of this issue that is probably fine since we already provide implementations of `printf` and friends there - maybe also the syslog function

PR #2183 (2024-09-19): Move binary logging to sys crate
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] I ha [... cut]
closed 2024-09-19 by jessebraham