esp-rs/esp-hal #1950: How to `Exclusive SPI Bus`

notBroman, 2024-08-14:
Heya,
I'm trying to use the spi bus of an ESP32 (`spi::master::Spi`) via `blocking::spi::SpiDevice` in `embassy_embedded_hal` as suggested by the [SPI documentation](https://docs.esp-rs.org/esp-hal/esp-hal/0.19.0/esp32/esp_hal/spi/master/index.html) to drive a waveshare EPD, but I am struggling to get it to compile. The program has been cut down to be as simple as possible in my debugging efforts.  I've tried searching the issues and other github repos to see how it's supposed to be done, but I can't find anything that shows how to use these two in tandem. 

Please help me resolve the error.

Cargo.toml
```rust
[dependencies]
esp-backtrace = { version = "0.11.0", features = [
    "esp32",
    "exception-handler",
    "panic-handler",
    "println",
] }
esp-hal = { version = "0.17.0", features = [ "esp32", "embedded-hal"] }
esp-println = { version = "0.9.0", features = ["esp32", "log"] }
log = { version = "0.4.20" }
esp-alloc = { version = "0.3.0" }
embedded-svc = { version = "0.26.1", default-features = false, features = [] }
embedded-io = "0.6.1"
esp-wifi = { version = "0.5.0", features = [
    "esp32",
    "phy-enable-usb",
    "utils",
    "wifi",
] }
heapless = { version = "0.8.0", default-features = false }
smoltcp = { version = "0.11.0", default-features = false, features = [
    "medium-ethernet",
    "proto-dhcpv4",
    "proto-igmp",
    "proto-ipv4",
    "socket-dhcpv4",
    "socket-icmp",
    "socket-raw",
    "socket-tcp",
    "socket-udp",
] }
epd-waveshare = "0.5.0"
embedded-hal = "1.0.0"
embassy-embedded-hal = "0.1.0"
embedded-hal-bus = "0.2.0"
embedded-graphics = "0.8.1"
embassy-sync = "0.6.0"
static_cell = "2.1.0"
[profile.dev]
opt-level = "z"
debug = true

[profile.release]
codegen-units = 1 # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false
```

The code
```rust
#![no_std]
#![no_main]

use esp_backtrace as _;

use embedded_hal::delay::DelayNs;

use esp_hal::{
    clock::ClockControl, 
    delay::Delay,
    gpio::{IO, NO_PIN},
    peripherals::Peripherals, 
    prelude::*, 
    spi::{master::Spi, SpiMode},
};

use epd_waveshare::prelude::*;
use epd_waveshare::epd1in54b::{Epd1in54b, Display1in54b, *};

use embassy_sync::blocking_mutex::Mutex;
use embassy_embedded_hal::shared_bus::blocking::spi::SpiDevice;

use core::cell::RefCell;

#[entry]
fn main() -> ! {
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();

    let mut clocks = ClockControl::max(system.clock_control).freeze();
    let mut d = Delay::new(&clocks);

    esp_println::logger::init_logger_from_env();

    
    // add pin driver
    let mut io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let mut led = io.pins.gpio2.into_push_pull_output();

    // reset the spi bus
    //      LOW: reset the device
    let mut rst = io.pins.gpio26.into_push_pull_output();

    // D/C  How the incomming data is interpreted
    //      HIGH: data
    //      LOW: command
    let mut dc = io.pins.gpio25.into_push_pull_output();
    dc.set_high();

    // BUSY when the device should not be interrupted
    //      HIGH: BUSY
    let mut busy_in = io.pins.gpio27.into_pull_down_input();

    let sclk = io.pins.gpio14;
    let mosi = io.pins.gpio13;
    let cs = io.pins.gpio15.into_push_pull_output();

    let spi = Spi::new(
        peripherals.SPI2,
        100_u32.kHz(),
        SpiMode::Mode0,
        &clocks,
    )
    .with_sck(sclk)
    .with_mosi(mosi);

    let spi = Mutex::new(RefCell::new(spi));
    let spi_dev = SpiDevice::new(& spi, cs);

    // Setup the epd
    let mut epd = Epd1in54b::new(&mut spi_dev,
                        busy_in.into(), 
                        dc.into(), 
                        rst.into(),
                        NO_PIN.into(),
                        &mut d.into()).expect("eink initalize error");

    log::info!("Hello world!");
    loop {

        led.toggle();
        d.delay_ms(1000);

    }
}
```
The error

```zsh
  --> src/main.rs:13:43
   |
13 | use epd_waveshare::epd1in54b::{Epd1in54b, Display1in54b, *};
   |                                           ^^^^^^^^^^^^^  ^
   |
   = note: `#[warn(unused_imports)]` on by default

error[E0308]: mismatched types
  --> src/main.rs:64:34
   |
64 |     let spi_dev = SpiDevice::new(& spi, cs);
   |                   -------------- ^^^^^ expected `&Mutex<_, RefCell<_>>`, found `&Mutex<_, RefCell<Spi<'_, SPI2, FullDuplexMode>>>`
   |                   |
   |                   arguments to this function are incorrect
   |
   = note: `Mutex<_, RefCell<Spi<'_, SPI2, FullDuplexMode>>>` and `embassy_sync::blocking_mutex::Mutex<_, RefCell<_>>` have similar names, but are actually distinct types
note: `Mutex<_, RefCell<Spi<'_, SPI2, FullDuplexMode>>>` is defined in crate `embassy_sync`
  --> /Users/notbroman/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-sync-0.6.0/src/blocking_mutex/mod.rs:25:1
   |
25 | pub struct Mutex<R, T: ?Sized> {
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: `embassy_sync::blocking_mutex::Mutex<_, RefCell<_>>` is defined in crate `embassy_sync`
  --> /Users/notbroman/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-sync-0.5.0/src/blocking_mutex/mod.rs:25:1
   |
25 | pub struct Mutex<R, T: ?Sized> {
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: perhaps two different versions of crate `embassy_sync` are being used?
note: associated function defined here
  --> /Users/notbroman/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-embedded-hal-0.1.0/src/shared_bus/blocking/spi.rs:38:12
   |
38 |     pub fn new(bus: &'a Mutex<M, RefCell<BUS>>, cs: CS) -> Self {
   |            ^^^

For more information about this error, try `rustc --explain E0308`.
```

ProfFan, 2024-08-15:
@notBroman Could you post your `cargo tree -d` result here?

notBroman, 2024-08-18:
@ProfFan Here you go, thank you for your help.
```
bitflags v1.3.2
└── smoltcp v0.11.0
    ├── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)
    └── esp-wifi v0.5.1
        └── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)

bitflags v2.5.0
└── esp-hal v0.17.0
    ├── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)
    └── esp-wifi v0.5.1 (*)

embassy-sync v0.5.0
└── embassy-embedded-hal v0.1.0
    └── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)

embassy-sync v0.6.0
└── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)

embedded-graphics-core v0.3.3
└── epd-waveshare v0.5.0
    └── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)

embedded-graphics-core v0.4.0
└── embedded-graphics v0.8.1
    └── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)

embedded-hal v0.2.7
├── embassy-embedded-hal v0.1.0 (*)
├── embassy-time v0.3.2
│   └── embassy-embedded-hal v0.1.0 (*)
├── epd-waveshare v0.5.0 (*)
└── esp-wifi v0.5.1 (*)

embedded-hal v1.0.0
├── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)
├── embassy-embedded-hal v0.1.0 (*)
├── embassy-time v0.3.2 (*)
├── embedded-hal-async v1.0.0
│   ├── embassy-embedded-hal v0.1.0 (*)
│   └── embassy-time v0.3.2 (*)
├── embedded-hal-bus v0.2.0
│   └── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)
├── embedded-hal-nb v1.0.0
│   └── esp-hal v0.17.0 (*)
└── esp-hal v0.17.0 (*)

hash32 v0.2.1
└── heapless v0.7.17
    └── embedded-svc v0.26.4
        └── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)

hash32 v0.3.1
└── heapless v0.8.0
    ├── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)
    ├── embassy-sync v0.5.0 (*)
    ├── embassy-sync v0.6.0 (*)
    ├── embassy-time v0.3.2 (*)
    ├── esp-wifi v0.5.1 (*)
    └── smoltcp v0.11.0 (*)

heapless v0.7.17 (*)

heapless v0.8.0 (*)

heck v0.4.1
├── enum-as-inner v0.4.0 (proc-macro)
│   └── core-isa-parser v0.2.0
│       [build-dependencies]
│       └── xtensa-lx-rt v0.16.0
│           └── esp-hal v0.17.0 (*)
├── strum_macros v0.24.3 (proc-macro)
│   └── core-isa-parser v0.2.0 (*)
└── toml-cfg v0.2.0 (proc-macro)
    └── esp-wifi v0.5.1 (*)
    [build-dependencies]
    └── esp-wifi v0.5.1 (*)

heck v0.5.0
└── strum_macros v0.26.3 (proc-macro)
    ├── strum v0.26.2
    │   └── esp-metadata v0.1.0
    │       [build-dependencies]
    │       └── esp-hal v0.17.0 (*)
    └── strum v0.26.2
        └── esp-hal v0.17.0 (*)

nb v0.1.3
└── embedded-hal v0.2.7 (*)

nb v1.1.0
├── embassy-embedded-hal v0.1.0 (*)
├── embedded-can v0.4.1
│   └── esp-hal v0.17.0 (*)
├── embedded-hal-nb v1.0.0 (*)
├── esp-hal v0.17.0 (*)
└── nb v0.1.3 (*)

no-std-net v0.5.0
└── embedded-svc v0.26.4 (*)

no-std-net v0.6.0
└── esp-wifi v0.5.1 (*)

serde v1.0.203
├── basic-toml v0.1.9
│   └── esp-metadata v0.1.0 (*)
│   [build-dependencies]
│   └── esp-hal v0.17.0 (*)
├── esp-metadata v0.1.0 (*)
├── minijinja v1.0.21
│   [build-dependencies]
│   └── xtensa-lx-rt v0.16.0 (*)
├── serde_spanned v0.6.6
│   ├── toml v0.8.13
│   │   └── toml-cfg v0.2.0 (proc-macro) (*)
│   └── toml_edit v0.22.13
│       └── toml v0.8.13 (*)
├── toml v0.8.13 (*)
├── toml-cfg v0.2.0 (proc-macro) (*)
├── toml_datetime v0.6.6
│   ├── toml v0.8.13 (*)
│   ├── toml_edit v0.21.1
│   │   └── proc-macro-crate v3.1.0
│   │       └── esp-hal-procmacros v0.10.0 (proc-macro)
│   │           └── esp-hal v0.17.0 (*)
│   └── toml_edit v0.22.13 (*)
└── toml_edit v0.22.13 (*)
[build-dependencies]
└── esp-hal v0.17.0 (*)

serde v1.0.203
└── embedded-svc v0.26.4 (*)

strum v0.24.1
└── core-isa-parser v0.2.0 (*)

strum v0.26.2 (*)

strum v0.26.2 (*)

strum_macros v0.24.3 (proc-macro) (*)

strum_macros v0.26.3 (proc-macro) (*)

syn v1.0.109
├── enum-as-inner v0.4.0 (proc-macro) (*)
├── proc-macro-error v1.0.4
│   └── esp-hal-procmacros v0.10.0 (proc-macro) (*)
└── strum_macros v0.24.3 (proc-macro) (*)

syn v2.0.66
├── darling_core v0.20.9
│   ├── darling v0.20.9
│   │   ├── enumset_derive v0.8.1 (proc-macro)
│   │   │   └── enumset v1.1.3
│   │   │       ├── embedded-svc v0.26.4 (*)
│   │   │       ├── esp-hal v0.17.0 (*)
│   │   │       └── esp-wifi v0.5.1 (*)
│   │   ├── esp-hal-procmacros v0.10.0 (proc-macro) (*)
│   │   └── xtensa-lx-rt-proc-macros v0.2.1 (proc-macro)
│   │       └── xtensa-lx-rt v0.16.0 (*)
│   └── darling_macro v0.20.9 (proc-macro)
│       └── darling v0.20.9 (*)
├── darling_macro v0.20.9 (proc-macro) (*)
├── enumset_derive v0.8.1 (proc-macro) (*)
├── esp-build v0.1.0 (proc-macro)
│   [build-dependencies]
│   └── esp-hal v0.17.0 (*)
├── esp-hal-procmacros v0.10.0 (proc-macro) (*)
├── num-derive v0.4.2 (proc-macro)
│   └── esp-wifi v0.5.1 (*)
├── portable_atomic_enum_macros v0.2.1 (proc-macro)
│   └── portable_atomic_enum v0.3.1
│       └── esp-wifi v0.5.1 (*)
├── serde_derive v1.0.203 (proc-macro)
│   ├── serde v1.0.203 (*)
│   └── serde v1.0.203 (*)
├── strum_macros v0.26.3 (proc-macro) (*)
├── toml-cfg v0.2.0 (proc-macro) (*)
└── xtensa-lx-rt-proc-macros v0.2.1 (proc-macro) (*)

toml_edit v0.21.1 (*)

toml_edit v0.22.13 (*)

winnow v0.5.40
└── toml_edit v0.21.1 (*)

winnow v0.6.9
└── toml_edit v0.22.13 (*)

```

Would you mind talking me through your thought process? So I can tackle debugging myself, if/when I run into something similar in the future?

bugadani, 2024-08-18:
The compiler gives you a hint with these lines:

```
note: `Mutex<_, RefCell<Spi<'_, SPI2, FullDuplexMode>>>` and `embassy_sync::blocking_mutex::Mutex<_, RefCell<_>>` have similar names, but are actually distinct types
note: `Mutex<_, RefCell<Spi<'_, SPI2, FullDuplexMode>>>` is defined in crate `embassy_sync`
```

When you see errors like this, you can suspect that you accidentally use the wrong version of a dependency. In your case, you can see embassy-sync showing up twice in your dependency tree:

```
embassy-sync v0.5.0
└── embassy-embedded-hal v0.1.0
    └── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)

embassy-sync v0.6.0
└── blinky v0.1.0 (/Users/notbroman/Documents/projects/rust/blinky)
```

Try downgrading it to `0.5.0` in your `Cargo.toml`, and your issue may go away. Or it may change to a different one :)

notBroman, 2024-08-18:
Thank you, that solved the error.

closed 2024-08-18 by notBroman