esp-rs/esp-hal #1954: UsbSerialJtag async breaks probe-rs on C6 on latest git

dragonnn, 2024-08-15:
Hi! I update my project to latest git (to get embassy-executor match up) and my project now breaks debugging and flashing via probe-rs if I have enabled UsbSerialJtag.
This is what I get in probe-rs on first try (it flashes but after the program is running this happens:
```
 WARN probe_rs::session: Could not clear all hardware breakpoints: Probe(Usb(Custom { kind: Other, error: Stall }))
 WARN probe_rs::session: Failed to deconfigure device during shutdown: Probe(Usb(Custom { kind: Other, error: Stall }))
Error: An error with the usage of the probe occurred

Caused by:
    0: USB Communication Error
    1: endpoint STALL condition
  ```
This makes also the UsbSerial to stop outputting anything.
If I retry running probe-rs this is the output I get:
```
Error: Failed to open the debug probe.

Caused by:
    0: The debug probe could not be created.
    1: A USB error occurred.
    2: endpoint STALL condition
```
This works fine before I updated esp-hal. And my project still works fine if I disable UsbSerialJtag.
To add that UsbSerial does work fine if I connect an already flashed project and do not run probe-rs, I get some hello world output.
Any idea what could cause that?

MabezDev, 2024-08-16:
We've occasionally seen the USB stall in our HIL workflow, it's possible this isn't related to USJ at all, and could be a bug somewhere else in the stack. What probe-rs version are you running by the way?

cc: @bugadani any ideas?

bugadani, 2024-08-16:
Sorry, do I understand you correctly? Did this work correctly for you before you updated esp-hal?

closed 2024-08-16 by JurajSadel

dragonnn, 2024-08-16:
Thanks! That was quick, just tested it and works fine now ❤️

PR #1961 (2024-08-16): Do not reset `UsbSerialJtag` peripheral
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ [... cut]