esp-rs/esp-hal #1958: Support for active high CS in SPI

Tenkir, 2024-08-16:
I'm working with a [Sharp Memory Display](https://www.sharpsde.com/fileadmin/products/Displays/2016_SDE_App_Note_for_Memory_LCD_programming_V1.3.pdf), whos CS pin is active high. It does not seem like there is a way with the SPI module to specify this, and so I have had to manually manage the CS logic in my device driver.

Perhaps there is a way to do this that I didn't see looking through the DOCS.RS, but it would be helpful in these odd cases to have native support for active high CS.

Dominaezzz, 2024-08-16:
The hardware supports it but it's not exposed yet.

MabezDev, 2024-08-20:
It's also possible to create your own type that implements `embedded_hal::digital::OutputPin` and inverts the signal that way.

Closing for now.

closed 2024-08-20 by MabezDev

felipetavares, 2025-01-05:
How does this work in the latest version?

With `.21` I could  invert the pins with `.into_peripheral_output().inverted()`, with `.22` I think it has something to do with `esp_hal::gpio::interconnect::OutputSignal` but I can't figure out the API around that.