esp-rs/esp-hal #1966: embassy BLE example not showing up on macos on the esp32c3

Sycrosity, 2024-08-18:
When running the "wifi_embassy_ble" example on an esp32c3 on the most recent release (and also on the github main branch), the device is detectable and can be connected to on windows and android, but cannot be found on mac (having tried on two separate macos devices).

The logs incase they are useful:
```
I (24) boot: ESP-IDF v5.1.2-342-gbcf1645e44 2nd stage bootloader
I (24) boot: compile time Dec 12 2023 10:50:58
I (25) boot: chip revision: v0.4
I (29) boot.esp32c3: SPI Speed      : 40MHz
I (34) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 4MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 003f0000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=079e0h ( 31200) map
I (101) esp_image: segment 1: paddr=00017a08 vaddr=3fc84e5c size=00378h (   888) load
I (103) esp_image: segment 2: paddr=00017d88 vaddr=3fca47f4 size=00078h (   120) load
I (111) esp_image: segment 3: paddr=00017e08 vaddr=40380000 size=04e5ch ( 20060) load
I (125) esp_image: segment 4: paddr=0001cc6c vaddr=00000000 size=033ach ( 13228) 
I (131) esp_image: segment 5: paddr=00020020 vaddr=42000020 size=28b94h (166804) map
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
Connector created
Ok(CommandComplete { num_packets: 5, opcode: 3075, data: [0] })
Ok(CommandComplete { num_packets: 5, opcode: 8198, data: [0] })
Ok(CommandComplete { num_packets: 5, opcode: 8200, data: [0] })
Ok(CommandComplete { num_packets: 5, opcode: 8202, data: [0] })
started advertising
```

Nothing appears in wireguard with the esp extension, either.
This may be an issue with bleps, but I cannot verify this yet - that would be solved by #1933.

bjoernQ, 2024-08-19:
I don't have those but I remember Mac and especially iOS devices have their own idea how BLE peripherals should look and behave. What did you try to search for BLE devices? Without any 3rd party software, it's unlikely that MacOS will show anything at all

Sycrosity, 2024-08-19:
I attempted to search using only the default bluetooth macos gui - I will try and look into what macos requires (and if I get it working would it be worth me adding to an example?)

bjoernQ, 2024-08-19:
You might also want to try another BLE utility on MacOS - I think I used https://github.com/jnross/Bluetility before (long time since I used MacOS before)

Sycrosity, 2024-08-19:
that was completely the issue - I didn't understand the differences between bluetooth and BLE. Using the BLE scanner it appeared straight away (granted with a UUID service tag of "Health Thermometer", but I think thats just me messing around with UUIDs) - Thanks for the pointers @bjoernQ!

closed 2024-08-19 by Sycrosity