esp-rs/esp-hal #1986: Decide on ordering of Tx/Rx pairs in APIs

MabezDev, 2024-08-22:
We should decide on an order, and add it to the API guidelines, then check that all APIs that return or consume a tx/rx pair use that order.

bugadani, 2024-08-22:
Well, obviously `tx, rx` is the only right way to do it :)

Btw we might want to think about this a bit more generally, there are some inconsistent things in the HAL, like:

[image]
(timer, clocks, vs clocks, timer)

AnthonyGrondin, 2024-08-22:
`embassy-net` and `smoltcp` both seem to be using `rx, tx`.

MabezDev, 2024-08-27:
In our internal meeting we decided to try and be consistent with the ecosystem, we will use `rx, tx`.

SergioGasquez, 2024-09-05:
> Well, obviously `tx, rx` is the only right way to do it :)
>
> Btw we might want to think about this a bit more generally, there are some inconsistent things in the HAL, like: [...]

I guess after https://github.com/esp-rs/esp-hal/pull/1999 clocks/timer is no longer an issue

closed 2024-09-06 by SergioGasquez