esp-rs/esp-hal #1987: SW3 isn't correctly cfg'd away when it's used by embassy multicore executor

MabezDev, 2024-08-23:
https://github.com/esp-rs/esp-hal/blob/19db509403a3e403e28a530764b238c903dfeaa1/esp-hal/src/system.rs#L306-L307

The embassy feature no longer exists on esp-hal, therfore this cfg doesn't work correctly.

Dominaezzz, 2024-08-27:
Ideally this should be solved with the type system rather than a cargo feature.
Something along the lines of `esp_hal_embassy::init(&clocks, timer, software_interrupt3, .....)`.

One shouldn't need special code in the hal to add support for an async runtime.

bugadani, 2024-08-27:
While I agree with the idea of a type-system solution, I'm not entirely happy about the boilerplate necessary for this and I'd prefer re-introducing the feature instead. We can hide it from the users (depending on esp-hal-embassy could automatically enable the feature), and third party runtimes can do whatever they wish.

bugadani, 2024-08-27:
I believe it is worth optimising for the common use case. Third party runtimes can deal with pass-interrupt-by-value if they need to.

PR #2011 (2024-08-27): Re-add feature gate for software_interrupt3
Closes #1987
closed 2024-08-27 by bugadani