esp-rs/esp-hal #1991: Support USB External PHY and USB-SERIAL-JTAG coexistence

ProfFan, 2024-08-23:
This can be achieved per the TRM (page 942).

The first part would eliminate the need to enable the internal PHY alongside with Wi-Fi which apparently is not recommended per the manual.

The second part would allow the USB and USB-JTAG to be used concurrently, which allows easy, debuggable development of USB peripherals.

MabezDev, 2024-08-23:
This would be nice, something I wish I did for my keyboard. I don't think it should be that hard, but I don't know how to test it. Do you know of any devkits that use an external USB phy?

ProfFan, 2024-08-23:
@MabezDev I don't know, but [this](https://github.com/espressif/esp-idf/issues/9366#issuecomment-1191278722) seems to indicate a viable solution. I'll see if I can make something that has it and potentially ship to you guys?

Dominaezzz, 2025-04-18:
Any luck @ProfFan ? I haven't found any devkits or boards with an external PHY.

ProfFan, 2025-04-18:
Hi @Dominaezzz I haven't got started yet on the HW design yet, was very busy

Dominaezzz, 2025-05-23:
Whilst I haven't gotten it to work yet, I just wanted to point out the existence of https://www.adafruit.com/product/1280 which can accept a TUSB1106 that has been reported to work with the S3.