esp-rs/esp-hal #2027: Atomic operations in PSRAM (esp. with 3rd party libraries)

ProfFan, 2024-08-28:
AFAIK the current atomic operations on the S3 uses native atomics. However this creates a problem: a 3rd party library could be using atomics on their data structure placed on the SPI PSRAM.

In the IDF, they handled it by first checking for PSRAM address, then try acquire the external RAM lock, then do CAS: https://github.com/espressif/esp-idf/blob/c9df77efbf871d4c3ae9fb828778ff8c4ab36804/components/esp_hw_support/cpu.c#L199

I wonder if a similar thing should be done for the S3 (for example, a PR to `portable-atomic` that adds a feature for a similar implementation?

Based on discussions with @ivmarkov

bugadani, 2024-08-29:
I don't see a simple way out of this without introducing our own, "PSRAM-safe" atomics. Which will be a usability and interoperability nightmare, because library developers will not likely use it, or even know about it.

Turning portable-atomic into a shim of sorts for us is at least technically possible, but still doesn't solve every issue - for MCUs that can use the core atomic types, those ones can't really be shimmed for us to provide our own implementation.

I don't think we can extend portable-atomic to know about our particular memory mappings.

MabezDev, 2024-08-29:
Do we know if an exception is raised whilst accessing atomics in psram, or does it just silently fail? If we can detect it, that at the very least opens up some options for us.

ivmarkov, 2024-08-30:
> Do we know if an exception is raised whilst accessing atomics in psram, or does it just silently fail? If we can detect it, that at the very least opens up some options for us.

I doubt any exception is raised. From the APP/PRO CPU POV, accessing PSRAM is like accessing regular internal RAM. And it ends up accessing regular internal RAM after all - thanks to the caching MMU.

With that said, I'm not in the position to discuss a possible fix, as I do not understand the operation of the MMUs to a deeper-enough detail. All I know is that ESP-IDF provides a special xchg atomic implemented in software, which is to be used when PSRAM is enabled, as pointed to by @ProfFan .

With that said, whoever wrote and maintains [this](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/soc/esp32/psram.rs) is probably in that position. (@bjoernQ - is that you?).

I would be happy to know though why hardware atomics cannot work with PSRAM:
- Is it because the two CPUs might have their own separate dram caches for the same psram; why would we even want such a setup?
- Or is it because an atomic might span across two cache-lines (the end of one, and the beginning of the other) and as such the writing might actually not be atomic at all?

Anyway, I guess me and @ProfFan are just raising the flag that there is an issue with atomics and PSRAM, and somebody needs to look into it. For both ESP-IDF and baremetal by the way, as the ESP-IDF multicore targets are also happily using only hardware atomics ATM; and once we know of a fix, we should probably change the definition of the upstream multicore STD targets from just using hardware atomics to emitting calls to the atomic intrinsic functions, so that the user code can use whatever ESP-IDF had put in the definition of these intrinsic functions, depending on user settings).

Ah, one more question, couldn't resist:
- Would hardware atomics work correctly, if the programmer is ensuring that the relevant PSRAM memory region is accessed only by **one** of the two CPUs? As in, do we even have a workaround for the current situation...

bjoernQ, 2024-08-30:
> With that said, whoever wrote and maintains [this](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/soc/esp32/psram.rs) is probably in that position. (@bjoernQ - is that you?).

Yes, I ported the code and no I don't know. I also doubt an exception is raised there but we can just give it a try

I'm also curious about the "why doesn't it just work"

MabezDev, 2024-08-30:
I'll try and find those answers and get back to you.

In the meantime, the only truly bullet-proof way to solve this right now is to have a target variant where we disable hardware atomics, falling back to portable-atomic or something else to ensure atomicity (note: this is exactly what esp-idf does, when the work around is enabled). I'd really like to avoid doing this though if we can help it.

ivmarkov, 2024-08-30:
> I'll try and find those answers and get back to you.
>
> In the meantime, the only truly bullet-proof way to solve this right now is to have a target variant where we disable hardware atomics, falling [...]

The question is, is `portable-atomic` even having a feature that would enable atomics (at least on bare-metal, for ESP-IDF we have other options) to operate correctly.

The [critical-section](https://github.com/taiki-e/portable-atomic?tab=readme-ov-file#optional-features-critical-section) one is tempting, but do you remember why our support for `u64` on multicore riscv got kicked out from upstream STD? Because it essentially used disable-all-interrupts-critical-sections-plus-spinlocks and they said using spinloocks for atomic operations defeats their very purpose in the first place.

Also if you notice, the software PSRAM-safe xchg impl in the ESP-IDF does something different... yet similar. They seem to implement the "fallible" xchg variant only, which can sporadically fail. But then the question is, what would the user do when presented with such an xchg impl, other than re-trying in a loop, which suspiciously looks like a cousin to spinlocks anyway...

ProfFan, 2024-08-30:
> * Would hardware atomics work correctly, if the programmer is ensuring that the relevant PSRAM memory region is accessed only by **one** of the two CPUs?

I think the answer currently for the S3 is most likely true but I am not sure. It's probably good to open an issue in `esp-idf` and tag @igrr for help 🍭 

> for MCUs that can use the core atomic types, those ones can't really be shimmed for us to provide our own implementation.

If we have a custom target (for example `xtensa-none-elf-esp32s3`) this can be put as a part in `core` right? As we build core with the application anyway.

> The [critical-section](https://github.com/taiki-e/portable-atomic?tab=readme-ov-file#optional-features-critical-section) one is tempting, but do you remember why our support for `u64` on multicore [...]

I wonder what they did for Cortex-M0/M0+ and such? See https://doc.rust-lang.org/std/sync/atomic/index.html#portability
thumbv6m does not support CAS at all.

ivmarkov, 2024-08-30:
> I wonder what they did for Cortex-M0/M0+ and such? See https://doc.rust-lang.org/std/sync/atomic/index.html#portability thumbv6m does not support CAS at all.

If Cortex-M0/M0+ is always single-core (is it?) then disable-all-interrupts-one-way-or-another as a strategy does work fine and since it is a single core, you don't have the "is a spinlock a lock or just waiting" issue.

By the way... by reading the opening paragraph in the "Portability section"... namely:
"
All atomic types in this module are guaranteed to be [lock-free](https://en.wikipedia.org/wiki/Non-blocking_algorithm) if they’re available. This means they don’t internally acquire a global mutex. Atomic types and operations are not guaranteed to be wait-free. This means that operations like fetch_or may be implemented with a compare-and-swap loop.
"

I'm now even more puzzled as to why they kicked out our `u64` multicore support...
I mean, if we talk about "spinlocks" (that we used in that implementation...)
- Are these "locks" in the terminology of that section? <-- I DOUBT this now
- Or are these "waits" in the terminology of that section <-- I now believe this is the case, and if so, **our spinlocking impl should've not been kicked out in the first place**...

bugadani, 2024-08-30:
> If Cortex-M0/M0+ is always single-core (is it?) 

No, see RP2040

ivmarkov, 2024-08-30:
> > If Cortex-M0/M0+ is always single-core (is it?)
> 
> No, see RP2040

Ah so you mean for rp2040 the use a spinlock+disable-all-interrupts software impl? If yes, then this only reinforces my suspicion, that our `u64` atomic impl Shall Be Re-Instated...

I guess at the time (and now still) I was not understanding the subject matter well enough to object their decision on technical ground...

ivmarkov, 2024-08-30:
...and if spinlocking+disable-all-interrupts is fine as a strategy, then the `critical-section` impl of `portable-atomic` is good enough of a workaround for multicore.

ProfFan, 2024-08-30:
> No, see RP2040

Beaten me in 5 secs lol

> Ah so you mean for rp2040 they use a spinlock+disable-all-interrupts software impl? If yes, then this only reinforces my suspicion, that our `u64` atomic impl Shall Be Re-Instated...

Probably no, see https://github.com/rust-lang/rust/issues/99668

> ...and if spinlocking+disable-all-interrupts is fine as a strategy, then the `critical-section` impl of `portable-atomic` is good enough of a workaround for multicore.

For real-time applications like what we use ESPs for I think this is worse than the IDF way of "best effort" weak CAS, but can live without if it's too complicated to make something similar in Rust.

ivmarkov, 2024-08-30:
> > No, see RP2040
> 
> Beaten me in 5 secs lol
> 
> > Ah so you mean for rp2040 they use a spinlock+disable-all-interrupts software impl? If yes, then this only reinforces my suspicion, that our `u64` atomic impl Shall Be Re-Instated...
> 
> Probably no, see [rust-lang/rust#99668](https://github.com/rust-lang/rust/issues/99668)

TL;DR :( 
If you can summarize for me,  _what_ does anybody use for CAS in multicore M0?
Because I don't think - whatever is used - can avoid using a `loop` and a loop is as close to a spinlock as you can get? As both are essentially just wait-and-check OR wait-and-retry which is essentially the same?

> For real-time applications like what we use ESPs for I think this is worse than the IDF way of "best effort" weak CAS, but can live without if it's too complicated to make something similar in Rust.

Because the weak cas esp idf impl does not disable interrupts (which is... slow?) or if not why then?

Because a strong cas impl on top of a weak cas would _also_ require a loop anyway...

MabezDev, 2024-09-11:
> I would be happy to know though why hardware atomics cannot work with PSRAM

Whilst it's related to the cache/MMU setup, the reason it doesn't work isn't PSRAM is treated too much like flash cache. Unfortunately, the appropriate signals to provide the required locking (which aren't needed when caching flash) were also not implemented for caching PSRAM, meaning the CPU will happily do atomic modifications on PSRAM memory without actually ensuring it's an atomic operation.

As you mentioned, PSRAM is treated just like DRAM in the esp32 and esp32s3, and because of this, we can't use the `ATOMCTL` register to control what the CPU does when  it executes s32c1l on external memories.

> Would hardware atomics work correctly, if the programmer is ensuring that the relevant PSRAM memory region is accessed only by one of the two CPUs? As in, do we even have a workaround for the current situation

Hmm, I'm not sure if this would work, consider the case of running one core but sharing an atomic variable between an interrupt. I believe these would be two different accesses, but I suppose the CPU (being a single core in this case) can only do one thing at a time so access would be sequential?

So in summary, we have no way of detecting this, we need to be proactive when using PSRAM. Essentially for every atomic access, we need to 

- Check if the variable is located in PSRAM
- If it is, also acquire a global spinlock to ensure atomicity 

I guess the question is where we do this. I already suggested a different target as a nuclear option but we can probably do better. Maybe we can do better, perhaps with something like `-Ctarget-feature=-feature` to disable atomics and fall back to libcalls that have the work around in place when psram is being used in the program?

MabezDev, 2024-09-13:
From our discussion in the meeting, we need to determine whether enabling `portable-atomic/critical-section` even on targets with native atomics will use a critical section. If it does, we at least have a nice-ish work around for the issue as we can detect PSRAM use and enable this feature.

This is not ideal though as it incurs a cost to all atomics, even ones stored in internal memory. Longer term it would be better to have a solution like esp-idf's where it checks where the atomic is before deciding whether to guard access with a spinlock.

bugadani, 2024-09-13:
As we're using atomic compare_exchange in our multi-core critical section implementation, can we enable `portable-atomic/critical-section` without things blowing up?

MabezDev, 2024-09-13:
> As we're using atomic compare_exchange in our multi-core critical section implementation, can we enable portable-atomic/critical-section without things blowing up?

Ah I forgot about that, good catch. Fortunately, I think we can control this by using `core::sync::atomic::*` in that impl, and ensuring that the global spinlock gets placed in internal memory via a linker attribute.

taiki-e, 2024-10-06:
> we need to determine whether enabling `portable-atomic/critical-section` even on targets with native atomics will use a critical section.

`portable-atomic/critical-section` is ignored if the target has atomic (i.e., `cfg!(target_has_atomic = "ptr")` is true).

I'm open to adding a new cfg to adjust the behavior here, but do not intend to change the default or add a new Cargo feature because:
- Due to bad behavior of the cargo v1 resolver, other platform-specific dependencies enable this feature, causing build failures or silent performance degradation.
- Some people unconditionally enable the `portable-atomic/critical-section` feature without reading documentation, e.g., https://github.com/matklad/once_cell/issues/264.
- Not officially documented, but in the current portable-atomic, if `cfg!(target_has_atomic = "N")` is true, `portable_atomic::Atomic{I,U}N` is lock-free[^1], but changing this behavior risks introducing issues into the code that assumes it.

[^1]:  Except in the case of a compiler bug, such as https://github.com/rust-lang/rust/issues/117305, https://github.com/rust-lang/rust/issues/117306

MabezDev, 2024-10-08:
Thanks @taiki-e :heart:!

> I'm open to adding a new cfg to adjust the behavior here, but do not intend to change the default

I think this entirely reasonable, I have a couple of questions:

- Do you have something in mind for the impl, or would you like us to have a go and PR to portable-atomic?
- What do you think the name of the feature should be?
- Do you think there is any scope for a conditional check? For instance we don't need to use the critical section if the atomic is placed in internal memory, so it would be nice to run a small address range check prior before committing to acquiring the lock.

MabezDev, 2024-10-10:
> I think there are two options on what implementation to use:

It sounds like using the `critical-section` impl would be more straight forward I think, but if you have some of the scaffolding ready for the global sharded lock, then it might be worth trying that. Either of these approaches would be acceptable to us.

> And even two options on when to use it:

For our specific use case, the latter is more attractive to me. Note that we only need to do this on four targets, `xtensa-esp32-none-elf`, `xtensa-esp32-esp-idf`, `xtensa-esp32s3-none-elf`, `xtensa-esp32s3-esp-idf`. Are you suggesting special casing some address ranges for these targets, or do you think a more general solution of somehow supplying an address range to portable-atomic?

MabezDev, 2024-10-24:
> Is it safe to call interrupt disable instruction from user code in xtensa-*-esp-idf? For example, RISC-V and Arm usually restrict the use of such instructions in user mode. If the same is true for Xtensa esp-idf, there will be no option to use anything other than critical-section.

@ivmarkov Could you clarify this, the details are a bit fuzzy for me.

> I think this depends on the actual spec and guarantees. If it is something that is clearly guaranteed in the spec, then I guess you can hard code it, but if it is something like “vendor may change it in the future” or “different for each chip or firmware configuration” then we can't hard code the value into the portable-atomic code.

The silicon has a hardcoded address space for PSRAM, it will never change. It is dependant on the chip configuration (at time of production) this means, the esp32 and esp32-s3 have different address spaces for PSRAM so we'll need a solution that accounts for that (I believe we can achieve this with cfg's). Firmware configuration has no impact.

ProfFan, 2025-02-16:
A gentle ping here @ivmarkov @MabezDev 

On the IDF part, I believe IDF already provides some kind of critical section: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html#api-changes which can be used directly

ivmarkov, 2025-02-16:
> A gentle ping here [@ivmarkov](https://github.com/ivmarkov) [@MabezDev](https://github.com/MabezDev)
>
> On the IDF part, I believe IDF already provides some kind of critical section: [...]

Even if ESP IDF does provide a workaround (their own library lockful atomics when psram is enabled, in fact) Rust Upstream will not accept a lockful implementation, because `core::sync::atomic` are by definition lock-free and thus cannot be lowered to libcalls that implement lockful atomics (i.e. with a disable-all-interupts-on-the-current-core-plus-spinlock critical section on a multi-core cpu).

More details why here: https://github.com/rust-lang/rust/issues/117305#issuecomment-2322870701 and the following comments between me and @bjorn3

A short term solution is to do something in `portable-atomic` which seem to be OK with lockful atomic. 

In the long term, if you figure out a way how to do lockless atomics on esp-idf I'm all ears, as then we can bring the case back to upstream Rust...

ivmarkov, 2025-02-16:
> [@ivmarkov](https://github.com/ivmarkov) Could you clarify this, the details are a bit fuzzy for me.

To be honest I don't know. I do not know the xtensa architecture on assembly level. With that said, there is no "user mode" and "privileged mode" in ESP-IDF. Both the kernel (the FreeRtos task scheduler, to be precise) and the drivers and the "user" code all run with the same privileges without any memory or other protection in between. So I would be surprised if there are any restrictions.

With that said, I think it would be most straightforward to just use the `critical-section` crate, because with esp-idf this crate anyway delegates to the esp-idf critical-section C impl - which - I assume - is efficient as it gets.

ivmarkov, 2025-02-17:
> > > Is it safe to call interrupt disable instruction from user code in xtensa-*-esp-idf? For example, RISC-V and Arm usually restrict the use of such instructions in user mode. If the same is true for Xtensa esp-idf, there will be no option to use anything other than critical-section.
> > 
> > 
> > [@ivmarkov](https://github.com/ivmarkov) Could you clarify this, the details are a bit fuzzy for me.
> 
> To be honest I don't know. I do not know the xtensa architecture on assembly level. With that said, there is no "user mode" and "privileged mode" in ESP-IDF. Both the kernel (the FreeRtos task scheduler, to be precise) and the drivers and the "user" code all run with the same privileges without any memory or other protection in between. So I would be surprised if there are any restrictions.
> 
> With that said, I think it would be most straightforward to just use the `critical-section` crate, because with esp-idf this crate anyway delegates to the esp-idf critical-section C impl - which - I assume - is as efficient as it gets.

@taiki-e @MabezDev short update:

**On the short-term approach (do something in `portable-atomic`):**

FYI, seems we can freely call interrupt enable/disable instruction on xtensa. ESP-IDF does exactly that in its public API `esp_cpu_compare_and_set`: https://github.com/espressif/esp-idf/blob/0d6099ec533c4b647fb7a7b0b8942bc7aeb82f90/components/esp_hw_support/cpu.c#L209

However I still believe just relying on `critical-section` (with a `cfg` ot whatever) is easier to comprehend even if that results in lockful atomics.

**On the long-term approach (lockless atomics in `core::sync::atomic` Rust Upstream):**

The `esp_cpu_compare_and_set` fn is actually interesting, in that it seems to correctly implement atomic-CAS-with-possible-spurious-failures **without** spinlocking the other CPU? My interpretation is that the user needs to re-try calling it if it returns `false` but I think my observation remains? Basically the contract of [`compare_exchange_weak`](https://doc.rust-lang.org/core/sync/atomic/struct.AtomicU32.html#method.compare_exchange_weak) if I'm not mistaken?

(It is a bit bulky in that it relies on a trick where there is a hidden atomic placed in non-external RAM, where hardware atomics do work fine, but still.)

If indeed so, wouldn't that function (or a similar equivalent) be a possible basis for lockless atomics impl for all atomics?

ivmarkov, 2025-02-17:
> > it seems to correctly implement atomic-CAS-with-possible-spurious-failures **without** spinlocking the other CPU?
> 
> Yes but I think the Rust Devs will argue that if one thread calls this function and gets stuck, no other thread can make progress anymore.
> 
> Like what happens if you call that function on a memory location mapped to flash whilst the other core is writing to flash or something.

Shouldn't calling that function with a readonly/flash/invalid location panic or generate a trap anyway (I mean, its safe variant)?
FWIW isn't this problem also present with the current hardware atomics, if you call them on flash memory? Even if PSRAM is disabled of course?

Dominaezzz, 2025-02-17:
Yes it would generate a trap (illegal access or something) but that atomic bool flag will be set (and will never be unset) then no other thread will be able to lock again.

The problem doesn't really exist for atomic ops in flash memory, whilst it'll still generate a trap, nothing gets stuck.

> It is another question how you even imagine you would call a safe Rust variant of the current hardware atomics or these future atomics when they are in flash?

Ah true, I don't have a good answer for that. 
Or maybe if this cas code lives in flash and executes while the cache is disabled, which would generate a trap?

ivmarkov, 2025-02-17:
> Yes it would generate a trap (illegal access or something) but that atomic bool flag will be set (and will never be unset) then no other thread will be able to lock again.

But then unless you "catch" traps (why would you?) this trap would be end-of-story for your current firmware execution anyway, so I still do not see the problem.

I think it might be more productive if we park the questionable topic of "atomics in flash" for the moment, and by putting it aside rather think in the direction of:
1. Does the `esp_cpu_compare_and_set` indeed implement the [compare_exchange_weak](https://doc.rust-lang.org/core/sync/atomic/struct.AtomicU32.html#method.compare_exchange_weak) contract? It think it does, but maybe @taiki-e can weigh-in here?
2. Does `esp_cpu_compare_and_set` qualify as a "lockless" (putting aside "flash")? I think it does?

ivmarkov, 2025-02-17:
The one problem I still see is the "NMI" scenario of bjorn3:
- If e.g. Thread-1 did get the first cas operation on the "regular-dram" atomic successfully, but is now suddenly hit by a NMI storm, it will stay there in the middle of the function not being able to make progress and thus not being able to reach the second cas on the "regular-dram" atomic
- If - at the same time - Thread-2 tries to do cas on the "regular-dram" atomic, it will fail with a "spurious failure", and will keep re-trying (if that's e.g. inside the implementation of - say - the PSRAM-compatible-"atomic-store")  and will thus make no progress.

In fact the whole `esp_cpu_compare_and_set` impl now looks to me suspiciously similar to a spinlock...

So maybe the question is a bit - is this "NMI storm" case possible to happen on Espressif xtensa chips? And if yes, what would the root cause of such a storm?

ivmarkov, 2025-02-17:
Oh well, as per [this discussion](https://matrix.to/#/!BHcierreUuwCMxVqOf:matrix.org/$IHKjXNa-5ZEzJDZhdwOQeZnYNyn3z9mGQUqiV5ZhQBU?via=matrix.org&via=catircservices.org&via=tchncs.de) disable-all-interrupts+spinlock (and likely `esp_cpu_compare_and_set`) are TECHNICALLY lockful in the end, so our only chance is to argue upstream that they are only lockful in the presence of "relatively abnormal" things happening (CPU holding the "cas" lock being shut-down), and that practically that might not be an issue.

But this also affects:
- Baremetal xtensa targets, which do have `core:sync::atomic` right now as well
- rp23, which has the exact same issue with PSRAM as esp32 xtensa MCU line, and also has `core::sync::atomic` which work with internal RAM only...

taiki-e, 2025-04-07:
@ivmarkov

Sorry for the late reply.

---

> @taiki-e @MabezDev short update:
>
> **On the short-term approach (do something in `portable-atomic`):**
>
> FYI, seems we can freely call interrupt enable/disable instruction on xtensa. ESP-IDF does [...]

I re-read the ISA documentation and found the following statement
(I don't recall seeing this in my earlier reading, so it may have been added recently.)

https://www.cadence.com/content/dam/cadence-www/global/en_US/documents/tools/silicon-solutions/compute-ip/isa-summary.pdf#page=582

> RSIL is a privileged instruction.

> Exceptions
> • EveryInstR Group (see EveryInstR Group)
> • Privileged Instruction Group (see Privileged Instruction Group)

If I understand correctly, this means that disabling instruction is a privileged operation, just like RISC-V and Arm, and the OS can raise an exception on the use of that instruction in user space if it wishes.

Therefore, I think the correct option here is to use the ESP-IDF's C API via critical-section, independent of whether the current ESP-IDF raises that exception or not.

(Also note that the fact that esp_hw_support uses privileged instructions does not necessarily mean that other user applications can do the same, since it is possible to only allow calls from certain blessed libraries and force the use of the C API for everything else, like [OpenBSD does for syscalls](https://github.com/golang/go/issues/36435).)

---

> * Does the `esp_cpu_compare_and_set` indeed implement the [compare_exchange_weak](https://doc.rust-lang.org/core/sync/atomic/struct.AtomicU32.html#method.compare_exchange_weak) contract? It think it does, but maybe @taiki-e can weigh-in here?

No.

compare_exchange_weak acquires a value, compares it to a user-specified value, attempts to exchange if equivalent, and returns the result of whether or not the value was actually exchanged and **the previous value**.

esp_cpu_compare_and_set is a function that acquires the global lock flag and returns false if another core has already acquired the lock, otherwise it compares and exchanges. This is different from compare_exchange_weak because it cannot even acquire the previous value if another core has already acquired the lock. Acquiring locks is necessary to avoid data races, even if only to acquire the value.

esp_cpu_compare_and_set is also incompatible with a normal atomic load/store (which done in a way that does not acquire the lock) to the same memory location and is UB (data race) if used concurrently.

taiki-e, 2025-04-07:
https://github.com/espressif/esp-idf/blob/a25e7ab59ed197817d4a78e139220b2707481f67/components/esp_hw_support/cpu.c#L224-L228

> ```c
>         // Re-enable interrupts
>         __asm__ __volatile__ ("memw \n"
>                               "wsr %0, ps\n"
>                               :: "r"(intr_level));
>     } else
> ```

By the way, why is the synchronization instruction not used after WSR in esp_cpu_compare_and_set? I would think that it is possible that the caller of the function could call the affected instruction...

ISA documentation says:

https://www.cadence.com/content/dam/cadence-www/global/en_US/documents/tools/silicon-solutions/compute-ip/isa-summary.pdf#page=650

> The point at which WSR.* to certain registers affects subsequent instructions is not always defined (SAR and ACC are exceptions). In these cases, the Special Register Tables in Special Registers on page 272 explain how to ensure the effects are seen by a particular point in the instruction stream (typically involving the use of one of the ISYNC, RSYNC, ESYNC, or DSYNC instructions). A WSR.* followed by an RSR.* to the same register should be separated with an ESYNC to guarantee the value written is read back. A WSR.PS followed by RSIL also requires an ESYNC.

ivmarkov, 2025-04-13:
> [@ivmarkov](https://github.com/ivmarkov)
> 
> Sorry for the late reply.
> 
> > > > > Is it safe to call interrupt disable instruction from user code in xtensa-*-esp-idf? For example, RISC-V and Arm usually restrict the use of such instructions in user mode. If the same is true for Xtensa esp-idf, there will be no option to use anything other than critical-section.
> > > > 
> > > > 
> > > > [@ivmarkov](https://github.com/ivmarkov) Could you clarify this, the details are a bit fuzzy for me.
> > > 
> > > 
> > > To be honest I don't know. I do not know the xtensa architecture on assembly level. With that said, there is no "user mode" and "privileged mode" in ESP-IDF. Both the kernel (the FreeRtos task scheduler, to be precise) and the drivers and the "user" code all run with the same privileges without any memory or other protection in between. So I would be surprised if there are any restrictions.
> > > With that said, I think it would be most straightforward to just use the `critical-section` crate, because with esp-idf this crate anyway delegates to the esp-idf critical-section C impl - which - I assume - is as efficient as it gets.
> > 
> > 
> > [@taiki-e](https://github.com/taiki-e) [@MabezDev](https://github.com/MabezDev) short update:
> > **On the short-term approach (do something in `portable-atomic`):**
> > FYI, seems we can freely call interrupt enable/disable instruction on xtensa. ESP-IDF does exactly that in its public API `esp_cpu_compare_and_set`: https://github.com/espressif/esp-idf/blob/0d6099ec533c4b647fb7a7b0b8942bc7aeb82f90/components/esp_hw_support/cpu.c#L209
> > However I still believe just relying on `critical-section` (with a `cfg` ot whatever) is easier to comprehend even if that results in lockful atomics.
> 
> I re-read the ISA documentation and found the following statement (I don't recall seeing this in my earlier reading, so it may have been added recently.)
> 
> https://www.cadence.com/content/dam/cadence-www/global/en_US/documents/tools/silicon-solutions/compute-ip/isa-summary.pdf#page=582
> 
> > RSIL is a privileged instruction.
> 
> > Exceptions
> > • EveryInstR Group (see EveryInstR Group)
> > • Privileged Instruction Group (see Privileged Instruction Group)
> 
> If I understand correctly, this means that disabling interrupts is a privileged operation, just like RISC-V and Arm, and the OS can raise an exception on the use of that instruction in user space if it wishes.
> 
> Therefore, I think the correct option here is to use the ESP-IDF's C API via critical-section, independent of whether the current ESP-IDF raises that exception or not.
> 
> (Also note that the fact that esp_hw_support uses privileged instructions does not necessarily mean that other user applications can do the same, since it is possible to only allow calls from certain blessed libraries and force the use of the C API for everything else, like [OpenBSD does for syscalls](https://github.com/golang/go/issues/36435).)

@taiki-e 

I agree `portable-atomic` on ESP-IDF should ideally call the ESP IDF C "critical-section" API though not necessarily for the concerns you raised. More on that later.

But let's - for a moment - put ESP-IDF out of the equation. I'll get back to this further below. 
_Since a lot of time had passed, let me try to make a summary as to where we stand_:

#### 1) Rust bare-metal

For Rust bare-metal apps without any "OS" (this is typically stuff based on the Embassy crates) - which want to use PSRAM - we do need a mode of operation of `portable-atomic` where _all `Atomic*` types and their operations are emulated_. Even if the MCU _does_ provide `core::sync::Atomic*` types in `core`. As these simply do not work when placed in PSRAM. I think the critical thing here is indeed "all". I.e. not just emulate `Atomic64` as there is e.g. no native `core::sync::Atomic64` implementation on the Espressif MCUs, but emulate everything, unconditionally.

Ideally, this can be implemented by just calling into the `critical-section` crate.
How this mode of operation can be enforced in `portable-atomic` is - I guess - open for debate. If this cannot be done with a `critical-section-emulate-fully` or suchlike feature (which would be oh so much more convenient!) OK I guess we can start with a `rustc` `--cfg`.

By the way, this is necessary not just for certain Espressif esp32 chips, but also for the Raspberry Pi Pico (`rp2040`) which 
a) is a dual-core Cortex M0 arch
b) does have `core::sync::Atomic*`
c) These do **not** work when placed in PSRAM either (there was a recent discussion about that in the Embassy chat)

But I think we are all in the meantime in agreement that this ^^^ is needed, and I think this is a bit more pressing than solving the Rust-app-on-top-of-ESP-IDF issue.

Basically this we need:
> I'm open to adding a new cfg to adjust the behavior here, but do not intend to change the default or add a new Cargo feature because:
>
> Due to bad behavior of the cargo v1 resolver, other [...]
Footnotes

@MabezDev @JamesMunns would you agree? (James - pinging you specifically for the rp2040, as I don't know the thing too well.)

#### 2) ESP-IDF

(
A digression:
I don't think ESP-IDF has a notion of "userspace" and "kernelspace" at all. All code - FreeRTOS, drivers, MBedTLS etc. etc. is lumped together in a single address space, with a single privilege level. Everything can call everything and execute any instruction. @MabezDev Correct me if I'm wrong.
)
[... 1 lines cut ...]
[... cut ...]
... hidden behind another new feature or cfg flag. It would be then up to ESP-IDF (or whatever) to provide impls for those at link time.

> No.
>
> compare_exchange_weak acquires a value, compares it to a user-specified value, attempts to exchange if equivalent, and returns the result of whether or not the value was actually exchanged and [...]

I completely agree. Hopefully, this point is irrelevant now though in light of my previous comment above about the introduction of the `extern Rust` pair of functions.

ivmarkov, 2025-04-15:
> One note, the RP2040 (M0+, no atomics) doesn't support PSRAM directly, but the RP235x (M33, does have atomics) does. The issue is that the system DOES have "native" atomics, but they _only_ work in the internal SRAM space, and NOT in memory-mapped PSRAM.

Thanks for correcting. I also corrected my original comment to refer to the RP Pico 2 instread.

> I haven't wrapped my head around [@ivmarkov](https://github.com/ivmarkov)'s proposed solution, but IMO I'm not sure if there actually is a viable path forward for targets like the rp235x, since users COULD use `core::sync::atomic::*` (or any library that uses it) if PSRAM space was exposed as a "normal" memory range or used for a general purpose allocator.

Certainly.

PSRAM-enabled chips like esp32 and esp32s3 are in the same boat. They **do** have `core::sync::atomic::*` Atomics that do NOT work with PSRAM. 

Yet, I would speculate most non-HAL-specific no_std code out there should and does anyway rely on `portable-atomic` instead of `core::sync::atomic::*` or else it won't be portable to multiple archs in the first place (as some targets don't have `core::sync::atomic::*` to begin with).

As for the rp235x and Espressif HALs themselves - yes - if they do have `core::sync::atomic::*` usage inside their guts, in data structures that otherwise CAN live in PSRAM memory (e.g. non-DMA buffers and whatnot) these need to be migrated to `portable-atomic`.

But what other options do we have? We can (and probably should) try to push for "PSRAM-aware"-specific targets for rp235x, esp32 and esp32s3 in Rust upstream but I thought that's a more difficult battle to win (due to the whole "these-atomics-would-be-theoretically-lock-ful-nono" argument, and maybe only do it once we at least have a workaround downstream in `portable-atomic`?

bugadani, 2026-01-12:
We would like to move forward with this issue.

There are currently two ideas we are considering:
- Custom JSON targets: if esp-hal detects that the user is trying to use PSRAM with the upstream target, it would panic. The user would need to support a custom target JSON that somehow(TM) makes the atomic operations safe.
- Changing the compiler: we can [disable atomic CAS](https://github.com/rust-lang/rust/blob/561364e4d5ccc506f610208a4989e91fdbdc8ca7/compiler/rustc_target/src/spec/targets/xtensa_esp32_none_elf.rs#L22), and force everyone to use portable_atomic instead. We either change portable-atomic to use critical sections __when it has to__, or a "libcall" - depending on which solution would be acceptable.

taiki-e, 2026-01-12:
I would be happy to accept the patch to implement https://github.com/taiki-e/portable-atomic/issues/215.

bugadani, 2026-01-12:
@taiki-e I'm worried about the performance implications of using a critical section where it's not necessary. What we would like is a conditional implementation: if the atomic is in a specific address range, branch to the critical-section fallback, but otherwise run the otherwise-available implementation that rustc would emit.

taiki-e, 2026-01-14:
IIUC which storage space doesn't support atomic operations depends on the device, so I think we need to provide a way to specify that. So, I think it would be good to require critical-section or unsafe-assume-privileged (which provides global-spinlock-based fallback that disable interrupts during its critical section) to be enabled simultaneously when that is specified.

bugadani, 2026-03-12:
Nominating for bug jury for bot feedback - this issue is being actively worked on, but takes time. I don't think it's likely to be finished in the next 45 days, so I'd like a way to signal that it's not abandoned, but not necessarily something I can provide a meaningful update for.

taiki-e, 2026-09-26:
> With [rust-lang/rust#158586](https://github.com/rust-lang/rust/pull/158586) it is now explicitly illegal to provide core::atomic for ESP32 and S3, so the next step will be to not wrap, but replace the core::atomic in portable-atomic, and to turn the feature off in the compiler. Whether that should be/will be done globally, or if we are going to have a -psram variant for them, I don't know, that part needs input from [@MabezDev](https://github.com/MabezDev) .

Since `std` requires that atomic CAS is available, it is doubtful whether atomic CAS can be disabled for ESP-IDF targets.

bugadani, 2026-10-22:
> Since std requires that atomic CAS is available, it is doubtful whether atomic CAS can be disabled for ESP-IDF targets.

We can do it for the bare metal targets still. Leaving this issue wide open is not a good situation, it's too easy to run into it accidentally.

As an alternative, we could provide two sets of targets, one with atomics enabled, one without. We they would have to place runtime checks into esp-hal's PSRAM setup code, that causes a panic on the atomics-enabled target. This still needs an assembly implementation in portable-atomic (I have it prepared, but not yet submitted), it still needs library changes to use portable-atomic more widely, it just lets people opt out of PSRAM via either a custom target, or a rustc-provided one.

Thinking this through, maybe we will need to do both, regardless of our preference.