esp-rs/esp-hal #2036: Support Sniffer-only mode in `esp-wifi`

ProfFan, 2024-08-29:
Rationale: `esp_wifi_80211_tx` has a lot of restrictions if the device has a Wi-Fi connection, per https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/wifi.html.

Currently the sniffer can only be enabled after AP/STA interface is already configured. It would be better if a dedicated sniffer mode can be supported.

CC @Frostie314159

MabezDev, 2024-09-16:
Is this actually possible with the available wifi blobs right now?

Frostie314159, 2024-09-16:
With the current blobs no. For various reasons, they require an interface to be active.
I did however manage to create a PoC without the blobs, but it's pretty hacky.

MabezDev, 2024-09-16:
Thanks for clarifying! I figured this might be the case. For many reasons we'll probably use the blobs here in esp-wifi for the foreseeable future so I'll close this issue for now. I'd love to see this work out in a 3rd party crate. FYI we just created https://github.com/esp-rs/esp-hal-community to home crates that don't quite belong here, but are also related.

closed 2024-09-16 by MabezDev