esp-rs/esp-hal #2050: esp_wifi::initialize disables floating point operations

Fundevoge, 2024-08-31:
Calling esp_wifi::initialize somehow disables floating point operations. This was not the case in an earlier version, although I do not remember which one.
```rs
println!("1 * 1 = {:.2}", 1.0_f32);

let wifi_init = esp_wifi::initialize(
    esp_wifi::EspWifiInitFor::Wifi,
    wifi_timer,
    Rng::new(peripherals.RNG),
    peripherals.RADIO_CLK,
    &clocks,
)
.unwrap();

println!("2 * 2 = {:.2}", 4.0_f32);
```

Leads to

```
1 * 1 = 1.00
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
2 * 2 =

Exception occured 'Cp0Disabled'
Context
PC=0x42016b1a       PS=0x00060730
0x42016b1a - core::fmt::float::<impl core::fmt::Display for f32>::fmt
```

bugadani, 2024-08-31:
You can add `xtensa-lx-rt` with the `float-save-restore` feature enabled and this error should go away.

Fundevoge, 2024-08-31:
Thanks, that works

closed 2024-08-31 by Fundevoge

plaes, 2024-09-01:
Could this be added to some kind of FAQ?