esp-rs/esp-hal #2052: [RFC] Upstream Wi-Fi registers.

Frostie314159, 2024-09-01:
Hi, I'm currently in the process of implementing a Wi-Fi stack for the ESP32 in pure rust and I have to use raw pointer to access the registers, we found while reversing.
I wanted to ask if we could upstream our patches for the SVD, to access them through `esp-hal`, or should maintain them out of tree.

MabezDev, 2024-09-02:
It should be fine to upstream your findings to the PACs, we're unlikely to use them in esp-wifi (unless you figured it all out :D) but I see no harm in them being there, provided they are of decent quality.

Frostie314159, 2024-09-02:
Yeah, they're probably not of much use for esp-wifi. The thing is, that most of these names are educated guesses at best, since they have like two references and some random value is written to them. Some are more obvious, like `0x3ff73090`, which is referenced in `hal_mac_rx_get_last_dscr`. I'd suggest putting them behind a feature flag.