esp-rs/esp-hal #2053: ESP32: wifi_embassy_esp_now does not work any more. `Exception occured: Cp0Disabled`

frederictobiasc, 2024-09-01:
Hi. Since v0.20.1, the [wifi_embassy_esp_now](https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/wifi_embassy_esp_now.rs) example fails with `Exception occured 'Cp0Disabled'`

```
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (31) boot: compile time Jun  7 2023 07:48:23
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0eb14h ( 60180) map
I (127) esp_image: segment 1: paddr=0001eb3c vaddr=3ffb0000 size=00e10h (  3600) load
I (129) esp_image: segment 2: paddr=0001f954 vaddr=3ffc3bf4 size=00190h (   400) load
I (134) esp_image: segment 3: paddr=0001faec vaddr=40080000 size=0052ch (  1324) load
I (142) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=4b8e8h (309480) map
I (262) esp_image: segment 5: paddr=0006b910 vaddr=4008052c size=0d128h ( 53544) load
I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }

Exception occured 'Cp0Disabled'
Context
PC=0x40107baa       PS=0x00060515
```

bugadani, 2024-09-01:
Please see https://github.com/esp-rs/esp-hal/issues/2050#issuecomment-2323010837

frederictobiasc, 2024-09-01:
To make it work, it's necessary to disable the `embassy-generic-timers` feature.

MabezDev, 2024-09-02:
With the release of xtensa-lx-rt@0.17.1, this should no longer be an issue unless you are using floats inside interrupt handlers (in which case please enable `xtensa-lx-rt/float-save-restore`).

closed 2024-09-02 by MabezDev

frederictobiasc, 2024-09-06:
@MabezDev I can confirm that the example works on master. Are you planning on doing a new patch release with the examples fixed?

MabezDev, 2024-09-07:
We've already released esp-hal@0.20.1, and xtensa-lx-rt@0.17.1, they should get picked up with a cargo update.