esp-rs/esp-hal #2054: ESP32 DAC Consine Wave Output

kbleeke, 2024-09-01:
The ESP32 is supposed to have a cosine wave generator that can output to the DACs as well. 

Is it already possible to use this from esp-hal or how complicated would it be to add?

references:

https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32/api-reference/peripherals/dac.html#continuous-wave-output-continuous-dma-mode

https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf 29.4.4

SergioGasquez, 2024-09-02:
Currently, cosine wave generator is not supported in esp-hal, to add support for it, feel free to take a look at https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/analog/dac.rs, the TRM section that you shared and the ESP-IDF source code.

kbleeke, 2024-09-03:
Thinking about it, I want to output raw audio on one of the DACs. It seems, I actually want to use the continuous mode with DMA for this. The bindings seem to be there in esp_idf_sys. I'll try to post the result as a PR when it works

kbleeke, 2024-09-03:
Also, I probably want to (among other things) stream audio via a2dp. This currently requires ESP-IDP, right? I don't know of any Rust BT stack that can do a2dp with e.g. embassy. 

Not sure what the relationship is, if I want to use the DAC from ESP-IDP, I need to add it to esp-idf-hal?

bjoernQ, 2024-09-03:
a2dp is classic Bluetooth which is only supported by the original ESP32 and not on bare metal - this is the repository of the bare metal HAL. You probably want to ask in https://github.com/esp-rs/esp-idf-hal

closed 2024-09-05 by kbleeke

Dominaezzz, 2025-04-30:
English is hard.

Only the original esp32 has the hardware to do Bluetooth classic.

no_std doesn't have software support for Bluetooth classic yet, which is needed for a2dp.

std might have some software support for it, check there to see if it's available.

KOEGlike, 2025-04-30:
Could I add support for a2dp in esp-hal?

KOEGlike, 2025-04-30:
should i open an issue first?