esp-rs/esp-hal #2059: Async SPI transaction with [write, read] operation broken on esp32

plaes, 2024-09-02:
I have [a driver](https://github.com/lora-rs/lora-rs/blob/a25e970225f4976a7b71d7159f7c69b5b0b70860/lora-phy/src/interface.rs#L54-L56) that does spi.transaction with sequential write and read operations.

```rust
let write_buffer = [0; 1];
let mut read_buffer = [0xa; 4];
let mut ops = [Operation::Write(write_buffer), Operation::Read(read_buffer)];
spi.transaction(&mut ops).await?;
```

It seems that whenever the `read_buffer` is longer than one byte, it doesn't get updated with read data, ie with this code `read_buffer` still contains `[0xa, 0xa, 0xa, 0xa]` after transaction.

Dominaezzz, 2024-09-02:
Thanks for the snippet, mind making it bigger? Is that with DMA? If so, how big are the buffers?
A full MVP would be appreciated then I'll be able to run it and see what's going on later this evening.

plaes, 2024-09-02:
> Thanks for the snippet, mind making it bigger? Is that with DMA? If so, how big are the buffers? A full MVP would be appreciated then I'll be able to run it and see what's going on later this evening.

Yes, with async fullduplex DMA. I'll see whether I can whip up something, as I was blocked by the xtensa-lx-rt issue.

Dominaezzz, 2024-09-02:
Try changing `let mut read: [u8; 2] = [0xff; 2];` to `let mut read: [u8; 2] = [0xab; 2];`.

I'm curious to see if the driver isn't writing anything to the buffer at all or if it's write 1s (from a pull up or something).

plaes, 2024-09-02:
Well, no pullup issue:
```
[INFO] - Test 1... result: [171, 171]
[INFO] - Test 2... result: [0]
[INFO] - Test 1... result: [171, 171]
```

plaes, 2024-09-02:
> Does the same happen for blocking mode?
I haven't yet figured out blocking + DMA + transaction, but [blocking + transaction](https://github.com/plaes/rust-lilygo-ttgo-lora32/blob/main/src/spi_blocking_transaction.rs) gives following output:
```
[INFO] - Test 1... result: [0, 0]
[INFO] - Test 2... result: [0]
[INFO] - Test 3... result: [0, 0]
```
```

Dominaezzz, 2024-09-02:
I've been able to reproduce it. (Without your full example I wouldn't have been able to, it only comes out in particular circumstances it seems)

Dominaezzz, 2024-09-02:
Basically, it seems once a write happens, reads stop working correctly.

Dominaezzz, 2024-09-02:
It appears to be a silicon bug in the base ESP32.
https://github.com/espressif/esp-idf/commit/366e4397e9dae9d93fe69ea9d389b5743295886f
(An ESP32 `#ifdef` was added later. https://github.com/espressif/esp-idf/commit/0c3653b1fd7151001143451d4aa95dbf15ee8506)

This stuff should really go in the errata.

Looks like the ESP32 will need special logic in the SPI driver.... :disappointed:

Dominaezzz, 2024-09-02:
To reproduce, take the `embassy_spi` example and change it from `transfer` to `write` + `read`.

bjoernQ, 2024-09-10:
I also see problems on ESP32-S2: there a `write` and `read` only succeeds once - then it fails

closed 2024-09-10 by JurajSadel