esp-rs/esp-hal #2076: ESP32-S3: Async SPI DMA transfers stall at some point

sourcebox, 2024-09-04:
This issue is opened by request of @bugadani at the Matrix chat.

While initially working, the call to `write_async().await` (and probably also other related functions) stalls at some point. It's not reproducable when this exactly happens.

HAL revision: `49e14b7ccb10dfe5b4b6ed561759372303beddb1`.

I use DMA channel 1 for the SPI transfers. Channel 0 is also used for I2S transfers.

Relevant code:
```rust
pub async fn spi_write_read(
    &mut self,
    write_buffer: &[u8],
    read_buffer: &mut [u8],
) -> Result<(), Error> {
    self.cs_pin.set_low();
    let result1 = self.spi.write_async(write_buffer).await;
    let result2 = self.spi.read_async(read_buffer).await;
    self.cs_pin.set_high();

    if result1.is_err() || result2.is_err() {
        return Err(Error::SpiTransferFailed);
    }

    Ok(())
}
```

bugadani, 2024-09-04:
I just want to add that https://github.com/esp-rs/esp-hal/pull/2065 was aiming at addressing this, but apparently the issue is somewhere else.

IvanLi-CN, 2024-09-04:
Following. I think I'm having a similar issue with ESP32-C3 after updating to v0.20.1.

Dominaezzz, 2024-09-06:
@sourcebox what other operations do you perform on the SPI? Or is that the only function that uses SPI in your project?

sourcebox, 2024-09-06:
SPI is only used for an external flash chip. When the firmware boots, a bunch of configuration data is read from the flash using `sequential-storage`. It's about 16 items that are read in a loop. Sometimes, it works perfectly, sometimes only a few items are read, then it stalls. As long as the read operations succeed, the returned data seem to be correct.

No write operations are done. By putting some log statements at various points, I was able to track down that in the shown function, the `write_async` is called, but nothing after it.

What's also strange is the fact, that replacing the `write_async/read_async` sequence with `transfer_asyc` always leads to return garbage data. So something seems to go really wrong in the whole process.

The full code of the flash driver is this:
```rust
//! Asynchronous driver for the SPI flash chip.

use embedded_storage_async::nor_flash::{
    ErrorType, NorFlash, NorFlashError, NorFlashErrorKind, ReadNorFlash,
};

use crate::hal::{
    dma::DmaChannel1,
    gpio::*,
    peripherals::SPI2,
    spi::{master::SpiDmaBus, FullDuplexMode},
    Async,
};

/// Size of a page in bytes.
pub const PAGE_SIZE: u32 = 256;

/// Size of a sector in bytes.
pub const SECTOR_SIZE: u32 = 4096;

/// Number of pages.
pub const NUM_PAGES: u32 = 2048;

/// Number of sectors.
pub const NUM_SECTORS: u32 = NUM_PAGES / 16;

/// Total capacity in bytes.
pub const CAPACITY: u32 = PAGE_SIZE * NUM_PAGES;

/// Type alias for the SPI peripheral driver.
type SpiDriver = SpiDmaBus<'static, SPI2, DmaChannel1, FullDuplexMode, Async>;

/// Type alias for CS pin.
type CsPin = Output<'static, Gpio33>;

/// SPI flash chip.
pub struct Flash {
    /// SPI driver.
    spi: SpiDriver,

    /// Chip-select pin.
    cs_pin: CsPin,
}

impl Flash {
    /// Returns a new instance.
    pub fn new(spi: SpiDriver, mut cs_pin: CsPin) -> Self {
        cs_pin.set_high();
        Self { spi, cs_pin }
    }

    async fn read_status_async(&mut self) -> Result<u8, Error> {
        let mut buffer: [u8; 2] = [0; 2];
        buffer[0] = commands::READ_STATUS_REGISTER_1;

        self.cs_pin.set_low();
        let result = self.spi.transfer_in_place_async(&mut buffer).await;
        self.cs_pin.set_high();

        if result.is_err() {
            return Err(Error::SpiTransferFailed);
        }
[... 296 lines cut ...]
    /// Read status register 1.
    pub const READ_STATUS_REGISTER_1: u8 = 0x05;

    /// Read data.
    pub const READ_DATA: u8 = 0x03;

    /// Enable writes.
    pub const WRITE_ENABLE: u8 = 0x06;

    /// Erase a sector.
    pub const SECTOR_ERASE: u8 = 0x20;

    /// Program a page.
    pub const PAGE_PROGRAM: u8 = 0x02;

    /// Read manufacturer and device ID.
    pub const MANUFACTURER_DEVICE_ID: u8 = 0x90;

    /// Read JEDEC ID.
    pub const JEDEC_ID: u8 = 0x9F;

    /// Read unique ID.
    pub const UNIQUE_ID: u8 = 0x4B;
}
```

bugadani, 2024-10-08:
@sourcebox can I ask you to please retest whether you still experience this issue?

MabezDev, 2024-10-08:
> It looks ok now, the transfer does not stall anymore.

That's great to hear! I'll close this now :).

closed 2024-10-08 by MabezDev