esp-rs/esp-hal #2081: RGB Interface Support for esp32-s3

jacobzlogar, 2024-09-04:
I would like to get an idea of what's needed to implement a RGB interface for the LCD_CAM module, i was told by @Dominaezzz that the DMA api's aren't currently powerful enough to support RGB mode. For context i'm attempting to use a [qualia esp32s3](https://www.adafruit.com/product/5800) board from adafruit that uses a RGB TTL connector

Dominaezzz, 2024-09-06:
Long story short, all the options available in https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/lcd/rgb_lcd.html should be supported by the hal. Either by having the implementation here or providing APIs for users to DIY it. (I much prefer the latter).

Dominaezzz, 2024-09-08:
We also need one of the maintainers to have a devkit that can run an example with it, like [ESP32-S3-LCD-EV-Board](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-lcd-ev-board/index.html).

Dominaezzz, 2024-10-02:
Now that #2242 (with the DMA buffer view feature) has landed, I can submit a PR for this.
All the (non PSRAM related) features needed for a decent no_std version of esp_lcd can now be implemented outside of esp-hal.
The XIP from PSRAM feature is still an important piece of the puzzle but that doesn't have to block the initial driver.

PR #2415 (2024-11-19): Introduce RGB/DPI driver
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] I have added [... cut]
closed 2024-11-19 by MabezDev